realsenseai/librealsense #10873: Point interpolation vs effective measured points

Jackarnd, 2022-09-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   {Win 10) | 
| Platform                                 | PC |
| SDK Version                            |  legacy / 2.0  | 
| Language                            |  C++                          | 
| Segment			|  Robot       | 

### Issue Description
The point cloud generated by the RealSense is very (very) noisy. To counter this noise I decided to average a set of different frames at each pixel coordinate. After 50 frames the average does not improve the noisyness level. Averaging 50 frames or 1000 frames gives the same result. 
So I looked at the point cloud generated by the RealSense in one frame and I realized the noise seemed periodic like _waves_ which leads me to believe it is by design in a low level of the system that I don't have access. The _waves_ make me think the RealSense actually measures n points and interpolated all the other points we see in the frame, which creates _waves_.

I verified that by taking frames of a wall at 0.5 meters, 1m, 2m, 3m. The frame at 0.5m is completely flat, with no noise, no wave. But the further I get the bigger the waves get (to a certain extent, it seems that there is a filter that contains at least a little bit these waves). This kinds of confirms that there are effective measured points and interpolated points in between.

The thing is I can imagine ways to get rid of all the noise if I only have the actual measured points. But I don't know how to differentiate the measured points and the interpolated ones...

So I have two questions:
1. How do I get rid of this intense noise?
2. If there is no good solution how do I tell the actual measured points from the interpolated ones?

My code roughly looks like that:
```
    // Start stream pipeline
    pipe.start();

    // let 30 frames pass for auto-exposure
    for (int f = 0; f < 30; f++) 
        frames = pipe.wait_for_frames();
    
    // loop to acquire the frames
    for (int i = 0; i < NbFrames ; i++)
    {
        // Wait for the next set of frames from the camera
         frames = pipe.wait_for_frames();

        // get depth frame
        auto depth = frames.get_depth_frame();
        
        // filtering 
        depth = temporal.process(depth), spatial.process(depth), Hole_filling.process(depth);

        // Generate the pointcloud and texture mappings
        points = pc.calculate(depth);

        // Récupération des vertices
        auto vertices = points.get_vertices(); // Vertices declaration
        
       /**** 
           Do things with the retrieved vertices
        ****/
    }

    // stop stream pipeline
    pipe.stop();
```

MartyG-RealSense, 2022-09-06:
Hi @Jackarnd   Yes, point cloud waviness can increase as the distance of the observed object / surface from the camera increases.  I recommend first reading the discussion about wavy points clouds at https://github.com/IntelRealSense/librealsense/issues/1375 which provides detailed advice from a RealSense team member and then exploring the links to related discussions at the bottom of that page.

Jackarnd, 2022-09-07:
Thank you this gives me materials towards fixing the solution.

closed 2022-09-07 by Jackarnd