realsenseai/librealsense #10877: How to get RGB actual exposure by using d415 in win10 ?

Longxiaoze, 2022-09-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (Open RealSense Viewer v2.25.0)-05.13.00.50 | 
| Operating System & Version |   Win 10   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  pyrealsense2                          | 
| Language                            |  python                          | 
| Segment			|  others                          | 

### Issue Description
I am using realsense d415 in win10. I want to get the actual exposure in meta data. But 'show frame metadata' of RGB Camera in inter realsense viewer v2.25.0 does not have 'Actual Exposure' of 'show frame metadata' of Stereo Camera.
I'm new to win10 and not sure if installing the registry can get the actual exposure of RGB? How do I get through python pyrealsense2 without installing the registry.
Thanks.
[image]

MartyG-RealSense, 2022-09-07:
Hi @Longxiaoze  Although actual exposure metadata can be retrieved for depth, RGB actual exposure metadata cannot be obtained whilst auto-exposure is enabled, as described at https://github.com/IntelRealSense/librealsense/issues/2549#issuecomment-431863920 by a RealSense team member.

Longxiaoze, 2022-09-08:
> Hi @Longxiaoze Although actual exposure metadata can be retrieved for depth, RGB actual exposure metadata cannot be obtained whilst auto-exposure is enabled, as described at [#2549 (comment)](https://github.com/IntelRealSense/librealsense/issues/2549#issuecomment-431863920) by a RealSense team member.

ok,thanks! I hope that the acquisition of the actual exposure value of rgb in auto exposure mode can be supported in the future.

closed 2022-09-08 by Longxiaoze