realsenseai/librealsense #10878: How to record the depth value in D455

JasonChanJY, 2022-09-07:
I want to record the depth value of depth stream. Is there any method to do that? THX.

MartyG-RealSense, 2022-09-07:
Hi @JasonChanJY  If you are able to use C++ language, a RealSense team member created an adapted version of the RealSense SDK's **rs-capture** example program that could write depth values to a **.csv** format file.  

https://github.com/dorodnic/librealsense/tree/text_sample/examples/capture

The amended code is highlighted at the link below.

https://github.com/dorodnic/librealsense/blob/text_sample/examples/capture/rs-capture.cpp#L15-L31

Alternatively, you could try recording a **bag** file (which is like a video file of camera data) in the RealSense Viewer tool and then converting the bag's data to another format such as csv using the SDK's C++ **rs-convert** tool.

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

If you prefer to use Python then you could save the depth to an **npy** file as an array of scaled matrices, like in the script at https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225

Another approach would be to save the depth as a **.raw** format image file that preserves depth information (which the RealSense Viewer tool's **Snapshot** option can do), and then retrieve depth from the .raw file using scripting such as the code at https://github.com/IntelRealSense/librealsense/issues/2231#issuecomment-416814331

JasonChanJY, 2022-09-07:
> Hi @JasonChanJY If you are able to use C++ language, a RealSense team member created an adapted version of the RealSense SDK's **rs-capture** example program that could write depth values to a **.csv** format file.
> 
> https://github.com/dorodnic/librealsense/tree/text_sample/examples/capture
> 
> The amended code is highlighted at the link below.
> 
> https://github.com/dorodnic/librealsense/blob/text_sample/examples/capture/rs-capture.cpp#L15-L31
> 
> Alternatively, you could try recording a **bag** file (which is like a video file of camera data) in the RealSense Viewer tool and then converting the bag's data to another format such as csv using the SDK's C++ **rs-convert** tool.
> 
> https://github.com/IntelRealSense/librealsense/tree/master/tools/convert
> 
> If you prefer to use Python then you could save the depth to an **npy** file as an array of scaled matrices, like in the script at [#4934 (comment)](https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225)
> 
> Another approach would be to save the depth as a **.raw** format image file that preserves depth information (which the RealSense Viewer's **Snapshot** option can do), and then retrieve depth from the .raw file using scripting such as the code at [#2231 (comment)](https://github.com/IntelRealSense/librealsense/issues/2231#issuecomment-416814331)

Is it possible to get the depth value from  .raw format image in Matlab?

MartyG-RealSense, 2022-09-07:
https://github.com/IntelRealSense/librealsense/issues/5824#issuecomment-584780985 offers advice about loading a RealSense .raw file into MATLAB.

JasonChanJY, 2022-09-07:
> [#5824 (comment)](https://github.com/IntelRealSense/librealsense/issues/5824#issuecomment-584780985) offers advice about loading a RealSense .raw file into MATLAB.

Thank you for your reply. I try to load .raw file into MATLAB successfully. And another question I want to ask that the index of the point as shown in the .png file means 1144 mm?

[image]

MartyG-RealSense, 2022-09-07:
It looks as though it refers not to mm depth but instead to the index value of a particular point on an **indexed image** in MATLAB, as discussed at the links below:

https://stackoverflow.com/questions/25853320/how-to-obtain-the-index-value-from-an-indexed-image-in-matlab

https://stackoverflow.com/questions/34256426/incorrect-pixel-value-from-data-cursor-on-matlab

JasonChanJY, 2022-09-07:
> It looks as though it refers not to mm depth but instead to the index value of a particular pixel on an **indexed image** in MATLAB, as discussed at the links below:
> 
> https://stackoverflow.com/questions/25853320/how-to-obtain-the-index-value-from-an-indexed-image-in-matlab
> 
> https://stackoverflow.com/questions/34256426/incorrect-pixel-value-from-data-cursor-on-matlab

Thanks. I check depth value in RealSense viewer, and the index value in MATLAB is consist with the depth value.

closed 2022-09-10 by JasonChanJY