realsenseai/librealsense #10886: temporal filter change depth image scope

qq996235826, 2022-09-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i  | 
| Firmware Version                   | Signed_Image_UVC_5_13_0_55.bin | 
| Operating System & Version |  Win 11  | 
| Platform                                 | PC  |
| SDK Version                            |   2.51.1                          | 
| Language                            |  python                         | 
| Segment			|  others                          | 

### Issue Description
I use pyrealsense2.temporal_filter to smooth my depth frame
but when I get the temp_filtered, I found it can't align to my color frame
What can I do to ensure that the filtered depth frame and the pre-filtered frame remain aligned?

`
import threading
import pyrealsense2 as rs
import numpy as np
import cv2
from camera import txt_writer

PHOTO_PATH = "photo/"

CAMERA_PHOTO_PATH = "photo/camera_photos/"

DEPTH_PHOTO_PATH = "photo/depth_photo/"

class Camera(threading.Thread):
    def __init__(self, ):
        threading.Thread.__init__(self)
        self.flag = True
        self.photo_flag = False

    def run(self):
        # 配置深度和颜色流
        # Configure depth and color flow
        pipeline = rs.pipeline()
        config = rs.config()

        # 获取设备产品线，用于设置支持分辨率
        # Get device product line for setting supported resolutions
        pipeline_wrapper = rs.pipeline_wrapper(pipeline)
        pipeline_profile = config.resolve(pipeline_wrapper)
        device = pipeline_profile.get_device()

        # 创建对齐对象（深度对齐颜色）
        # Create an alignment object (depth alignment color)
        align = rs.align(rs.stream.color)
        found_rgb = False
        for s in device.sensors:
            if s.get_info(rs.camera_info.name) == 'RGB Camera':
                found_rgb = True
                break
        if not found_rgb:
            print("The demo requires Depth camera with Color sensor")
            exit(0)

        # 深度参数
        # depth parameter
        config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
        # 颜色参数
        # color parameter
        config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

        # 时间滤波器
        # time filter
        temporal = rs.temporal_filter()
        
        # 时间滤波计数器
        # time filter counter
        i = 0
        
        # 时间滤波帧存储
        # Time-filtered frame storage
        frameList = []
        
        # 深度图着色器
        # depth map colorizer
        colorizer = rs.colorizer()

        # 开始流式传输
        # start
        pipeline.start(config)
        try:
            while self.flag:
                # 等待一对连贯的帧：深度和颜色
                # Wait for a pair of consecutive frames: depth and color
                frames = pipeline.wait_for_frames()
                
                # 对齐后再获取
                # get after alignment
                aligned_frames = align.process(frames)
                
                # 获得对齐后的深度和颜色图
                # Get the aligned depth and colorframe
                depth_frame = aligned_frames.get_depth_frame()
                color_frame = aligned_frames.get_color_frame()

                if not depth_frame or not color_frame:
                    continue
                
                # 将图像转换为 numpy 数组
                # convert frame to numpy array
                depth_image = np.asanyarray(depth_frame.get_data())
                color_image = np.asanyarray(color_frame.get_data())
                
                # 在深度图像上应用颜色图（图像必须首先转换为每像素 8 位）
                depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
                depth_colormap_dim = depth_colormap.shape
                color_colormap_dim = color_image.shape

                # 时间滤波存储
                # add depth_frame to filter
                if 0 < i < 10:
                    frameList.append(frames.get_depth_frame())
                    i += 1

                # 如果需要拍摄就保存图片
                # take photo
                if self.photo_flag:
                    # 调用专门的写入线程来进行写入操作
                    # write file
                    writer = txt_writer.TxtPointWriter(depth_frame, color_frame, depth_image, color_image)
                    writer.start()
                    self.photo_flag = False
                    # 存入第一张时间滤波帧
                    # save the first depth frame to filter
                    frameList.append(frames.get_depth_frame())
                    i = 1

                # 写入时间存储,目前过滤10帧
                # filter 10 depth frame
                if i == 10:
                    i = -1
                    for x in range(10):
                        # 帧导入时间滤波器
                        # filter
                        temp_filtered = temporal.process(frameList[x])
                    # 获得深度图
                    # get the filtered image
                    depth_image = np.asanyarray(temp_filtered.get_data())
                    # 调用线程写入
                    # write file
                    writer1 = txt_writer.TxtPointWriter(temp_filtered, color_frame, depth_image, color_image,
                                                        'cloud1.txt')
                    print('写入时间滤波')
                    writer1.start()
                    # 深度图着色
                    colorized_depth = np.asanyarray(colorizer.colorize(temp_filtered).get_data())
                    unfilt_depth = np.asanyarray(colorizer.colorize(depth_frame).get_data())
                    cv2.imwrite(DEPTH_PHOTO_PATH+'filtered-depth.png', colorized_depth)
                    cv2.imwrite(DEPTH_PHOTO_PATH+'unfilt-depth.png', unfilt_depth)

                # 如果深度和颜色分辨率不同，请调整彩色图像的大小以匹配深度图像以进行显示
                if depth_colormap_dim != color_colormap_dim:
                    resized_color_image = cv2.resize(color_image, dsize=(depth_colormap_dim[1], depth_colormap_dim[0]),
                                                     interpolation=cv2.INTER_AREA)
                    images = np.hstack((resized_color_image, depth_colormap))
                else:
                    images = np.hstack((color_image, depth_colormap))

                # 显示图像
                cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
                cv2.imshow('RealSense', images)
                cv2.waitKey(1)

        finally:
            # 停止流式传输
            pipeline.stop()
`
**filtered-depth:**
[image]
**unfiltered-depth:**
[image]
**colorimage:**
[image]

MartyG-RealSense, 2022-09-09:
Hello, it appears that you are applying alignment first and temporal filter post-processing second.  Intel recommend applying alignment **after** post-processing, otherwise distortion effects such as aliasing (jagged lines) may occur.

qq996235826, 2022-09-10:
> Hello, it appears that you are applying alignment first and temporal filter post-processing second. Intel recommend applying alignment **after** post-processing, otherwise distortion effects such as aliasing (jagged lines) may occur.

After modifying it according to your suggestion, the problem is solved, thank you very much!

closed 2022-09-10 by qq996235826

MartyG-RealSense, 2022-09-10:
It's excellent news that you were successful.  :)  Thanks very much for the update!