realsenseai/librealsense #10887: temporal filter after alignment does not work

omerbrandis, 2022-09-09:
Hello,

I've encountered a problem somewhat similar to this [issue](https://github.com/IntelRealSense/librealsense/issues/10886)
Only I'm using c++, and my result is that the temporal filter seems to have no effect at all on the depth image.
I've built rs 2.29.0 from source , slightly modified the example/rs-post-processing.cpp, I've added :
1. set high accuracy  ( filter works )
2.  added rgb stream ( filter works ) 
3. added alignment   ( filer is having no effect on the depth_frame)
[rs-post-processing.txt](https://github.com/IntelRealSense/librealsense/files/9535791/rs-post-processing.txt)

i've noticed MartyG-RealSense's reponse to 10886, 
and I'm asking:
1. where has "Intel recommend applying alignment after post-processing" ? ( in which document ?)
2. how can i do this in c++
( rs2::align.process expects a frameset, but frameset does not have setter methods or suitable constructor )

omerbrandis, 2022-09-09:
Hello MartyG,

Thanks again for trying to help.

1. regarding the recommendation to perform pp before alignment : 
"We recommend to apply post-processing filters before aligning depth to color to reduce aliasing. This is the way it implemented in ''realsense-viewer''s pointcloud view."
-> i was under the impression that the rs-viewer does not perform alignment.
regardless, can you point me at the code ? ( I was not able to find it )

2. you mentioned that applying pp after alignment is not compulsory, 
which leads me back to my original question , why doesn't it work in the example i've provided?

( 3. regarding the various examples in #1658, 
it is my understanding that all of them do not work. )

thanks,
Omer.

MartyG-RealSense, 2022-09-10:
1.  The RealSense Viewer does not support alignment in 2D mode.  In 3D mode though it can map RGB onto depth points to generate a textured pointcloud.  My understanding is that this mapping is based on the **map.to and pc.calculate** method of producing a pointcloud, like in the C++ example program **rs-pointcloud**.

https://github.com/IntelRealSense/librealsense/blob/master/examples/pointcloud/rs-pointcloud.cpp

```
pc.map_to(color);
auto depth = frames.get_depth_frame();
// Generate the pointcloud and texture mappings
points = pc.calculate(depth);
```

When talking about the recommendation to align after post-processing, this typically refers to the **rs2::align** method of alignment, which is different from the map.to method above.  rs2::align makes use of the SDK's **align processing block** to make automatic adjustments for differences between the depth and color streams if there are differences between the two stream types regarding resolution or field of view size.

2.  In another recent case at https://github.com/IntelRealSense/librealsense/issues/10870#issuecomment-1239252467 where the temporal filter was apparently not working, I found that changes to the filter made no difference if only the temporal filter was being used but the image updated if a second type of post-processing filter was also being applied, such as Decimation or Spatial.

Intel's recommended order for applying post-processng filters at the documentation link below shows that Decimation and Spatial both come before Temporal in the list order, suggesting that if Decimation or Spatial is applied first in the list of filters in the script and Temporal applied second then the image may react to changes in the Temporal filter.

https://dev.intelrealsense.com/docs/post-processing-filters#using-filters-in-application-code

omerbrandis, 2022-09-10:
Hello Martyg,

thanks again for all your time and effort,
But I'm afraid i still do not understand how your comments answer my question.
I have demostrated that executing the temporal filter after aligment does not work as advertised ( it does not change the depth frame with accordance to the previous frames data )
there is nothing in the official documentation that prohibits this ( execution temporal filter after alignment ) .
if this were illegal/invalid , then the execution of temporal_filter::process should explicitly fail (/ throw an exception ).
so unless there's some sort of error in my demonstration ( rs-post-processing.txt) , then this is a bug in realsense ( v2.29.0).
Can you please forward this to development ?

MartyG-RealSense, 2022-09-10:
My comments above were suggesting enabling the Decimation or Spatial filter in your script (even if you do not need those filters) and then enabling the temporal filter secondly after that.  The purpose of this test would be to confirm whether the temporal filter requires another filter to first be enabled in order for the temporal filter to be able to update the image correctly. 

I would be happy to forward your query to Intel RealSense developers.  If a bug was present in SDK **2.29.0** though then a fix would not be able to be integrated into that version as the SDK is now up to version **2.51.1** and previous SDK versions cannot be retroactively edited and re-published, unfortunately.  Bug fixes can be introduced in **future** SDK versions but would not fix an issue in a version earlier than the one that the fix was implemented in.

Would it be possible to update your project's SDK version to 2.51.1 please?  If there was still an issue with the temporal filter in that latest release then it would then be practical to investigate the temporal filter for a possible bug.

MartyG-RealSense, 2022-09-10:
May I also ask which firmware version is currently installed in your camera?  It is recommend that firmware **5.11.15.0** or newer is used with 2.29.0.  However, the maximum version used with 2.29.0 should be **5.12.7.100**.  This is because from firmware 5.12.8.200 onwards, the minimum SDK version used should be **2.39.0** due to internal changes in the frmware from 5.12.8.200 onwards.

omerbrandis, 2022-09-10:
Hello MartG 

1. I'm running fm 5.11.15.00 on d435.
2. I am not interested in using the decimation and or Spatial filters at this time.
+ i've explained in the opening comment that the temporal filter stopped effecting the depth frame a lot after I removed the execution of the these filters.
3. I realize that a fix will only be applied to newer versions and that this will take some time.
In the mean time I have started looking into alternatives like executing the temporal filter before alignment
or forgoing alignment.
4. I  can not test this issue on the latest release at this time.
:-)

MartyG-RealSense, 2022-09-10:
I understand that you are not interested in the decimation or spatial filters.  Would it be possible for you to temporarily add one of them to your script anyway to provide me with diagnostic information about your problem, please. :)  I would like to confirm whether the temporal filter starts working in your program if you add a decimation or spatial filter.  

Thanks for your patience!

MartyG-RealSense, 2022-09-10:
If you are only using one frame but the temporal filter needs a history of multiple frames, would it be possible to use the method of the **rs-save-to-disk** C++ example, which captures 30 frames but only uses the 30th one.

https://github.com/IntelRealSense/librealsense/blob/master/examples/save-to-disk/rs-save-to-disk.cpp

omerbrandis, 2022-09-11:
Hello MartyG

file "rs-post-processing.txt" which demonstrates the problem is a modification of rs-post-processing.cpp that is included in librealsense's source distribution.
I first downloaded realsense source v 2.29.0, and built as is.
rs-post-processing.cpp includes the use of decimation_filter,threshold_filter,spatial_filter,
i tested the original version an it worked.  ( i guess i should/could have explicitly stated this as phase 0  in my initial comment ).
i then removed use of decimation_filter,threshold_filter,spatial_filter , temporal filter still worked ( phase 1 in my initial comment ).

MartyG-RealSense, 2022-09-11:
So the script in the **rs-post-processing.txt** file that you provided at the start of this discussion is a modification of rs-post-processing.  In the script's original version the temporal filter works.  After you have modified the script, the temporal filter doesn't work.  Is that correct, please?

A debug strategy that you could try if you have not done so already is to return to the original script and re-add your changes one at a time, then compile the script to see if the temporal filter still works and then add the next change in the script.  When the temporal filter stops working correctly then you will have likely found the cause of the problem.

***********

If you would prefer not to recreate your script from the beginning then the change that I would recommend trying first is to comment out the camera preset code that loads the High Accuracy preset to confirm whether or not the new configuration is disrupting the temporal filter.

omerbrandis, 2022-09-11:
i've not tested v2.50.0.
the error is not present there.
;-)

closed 2022-09-11 by omerbrandis