realsenseai/librealsense #10889: Ubuntu won't boot after installing

firatbozkaya, 2022-09-10:
I have an intel aero drone. Ubuntu won't boot after installing librealsense. It stays on the purple screen. I need help

MartyG-RealSense, 2022-09-10:
Hi @firatbozkaya   The Intel Aero Ready To Fly drone system, which is no longer supported, was originally designed to be used with a flavor of Linux called **Yocto Project** rather than Ubuntu.

https://github.com/intel-aero/meta-intel-aero-librealsense

However, Ubuntu **16.04** could be used with it.

https://www.intel.co.uk/content/www/uk/en/support/articles/000030952/drones/development-drones.html

The librealsense version that it refers to though is **librealsense1**, now known as Legacy Librealsense.  The librealsense1 SDK does not support modern RealSense camera models such as the 400 Series and L515 as it was designed with the old **R200** stereo camera model in mind (the original predecessor of the 400 Series stereo cameras).  

The current **librealsense2** SDK, also known as RealSense SDK 2.0, supports those modern models.

There is guidance about using Aero with modern D415 or D435 cameras at the link below.

https://docs.px4.io/v1.12/en/complete_vehicles/intel_aero.html#realsense-camera

The 'master branch' that the instructions suggest cloning is the latest version of the librealsense2 SDK.

https://github.com/IntelRealSense/librealsense

MartyG-RealSense, 2022-09-10:
If Ubuntu will not boot correctly, are you able to boot in **Recovery** mode by holding down the left shift or escape kry on the keyboard as the computer begins to start up?

https://www.maketecheasier.com/boot-recovery-mode-ubuntu/

closed 2022-09-23 by MartyG-RealSense

MartyG-RealSense, 2026-02-18:
There are a couple of other installation guides for Intel Aero that you could look at to see if they provide any new inspirations.

https://github.com/intel-aero/meta-intel-aero/wiki

https://docs.px4.io/v1.12/en/complete_vehicles/intel_aero.html