realsenseai/librealsense #10891: RealSense SDK & pyrealsense2 Installation on Nvidia Jetson Modules

35selim, 2022-09-12:
### Bash Script Written for Installation on Jetson Devices
Installing RealSense SDK and pyrealsense2 on a Jetson module that has ARM architecture is a bit confusing especially for new users. Even for users that has some experience with RealSense may have trouble while installing a fresh environment and this process may take several hours including online search. Therefore, I have written a bash script for the benefit of the RealSense community, so that one can easily initialize his/her Jetson device that has a fresh installation and can start utilizing RealSense SDK and pyrealsense2.

MartyG-RealSense, 2022-09-12:
Thanks so much @35selim for sharing your installation script with the RealSense community!  I have added a **Documentation** tag to this discussion so that it is kept open and also posted the link on a popular discussion about installing librealsense and the pyrealsense2 wrapper from source code on Jetson.  Thanks again!

densnz, 2022-10-12:
jetson xavier agx dev kit
Python 3.8.10 (default, Jun 22 2022, 20:18:18)
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.

import pyrealsense2
Traceback (most recent call last):
File "", line 1, in
ModuleNotFoundError: No module named 'pyrealsense2'

Your code helped me install the SDK, but pyrealsense2 could not install. Most likely the problem is in the PYTHONPATH, but it's not certain)

MartyG-RealSense, 2022-10-12:
Hi @densnz  If you are using the **build_pyrealsense2_and_SDK.sh** script of @35selim then that script sets the PYTHONPATH to Python 3.6.  So you may need to change python3.6 to **python3.8**

https://github.com/35selim/RealSense-Jetson/blob/main/build_pyrealsense2_and_SDK.sh#L23

You can find out what the PYTHONPATH is on your particular computer with the Python command **which python3**

densnz, 2022-10-12:
@MartyG-RealSense, I changed the file as you pointed out. But nothing has changed. Do I have to delete what I installed before and start over?

35selim, 2022-10-12:
Hi @densnz 
Downgrading your system's python version to 3.6 may help in your situation. Otherwise, consider changing your pythonpath code to 
```
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
```

densnz, 2022-10-12:
Hi @35selim,
Unfortunately, even this suggestion did not help my problem...

I downloaded an old version of python (3.6)
Installed it as the "default
Manually changed the .bashrc
I changed export $PYTHONPATH
No effect...

densnz, 2022-10-12:
> Hi again @densnz
> 
> Can you confirm that you didn't get any errors while running the script file and you can see a file starting with 'pyrealsense' and ending with '.so' in the file search (search at the root)?

The only error occurred at the very end of the build:
`cp: cannot stat '/home/densnz/.99-realsense-libusb.rules': No such file or directory`

( 'pyrealsense' not found )
`$ sudo find ~ -name pyrealsense2`
`/home/densnz/librealsense/wrappers/python/pyrealsense2`
`/home/densnz/librealsense/build/wrappers/python/CMakeFiles/Export/lib/cmake/pyrealsense2`

MartyG-RealSense, 2022-10-18:
Hi @35selim   Do you require further assistance with this case, please?  Thanks!

35selim, 2022-10-28:
Hi again. One user said [here](https://github.com/35selim/RealSense-Jetson/issues/1#issuecomment-1293894851) that Python binding problem is solved using the method. By the way, I don't have any further assistance but this issue is tagged as documentation by you @MartyG-RealSense, so it will stay open.

MartyG-RealSense, 2022-10-28:
Yes, the Documentation tag means that it will stay open.  I was checking whether you required further advice at this current time.  Thanks!

dannykha, 2023-01-31:
Hi @35selim I ran your scripts on my NVIDIA Jetson Xavier AGX running Ubuntu 20.04 and I have a problem with the last script, "sh install_SDK_without_pyrealsense2.sh" where it says 
E: Unable to locate package librealsense2-utils
Would you happen to know a solution to this problem? It seems it cant find the package from the repository. Thank you!

35selim, 2023-01-31:
Hi @dannykha 
The problem would be the same as stated [here](https://github.com/IntelRealSense/librealsense/issues/9040#issuecomment-847442449). I didn't make a deep search about whether new librealsense releases support Linux 20.04 distributions for ARM based devices or not, but it is possible to achieve a successful installation with downgrading the Linux version to, for example, 18.04. 
Can you please try this out?
Regards.

GaoQi-2023, 2023-10-02:
@35selim Hello! 
Sorry, I do not know what is happening but it is still "No module named pyrealsense2". 
My Jetson module is Jetson Nano 4G (not Jetson Orin Nano). I downloaded the JetPack SDK 4.6 today and ran your script soon after installing the image. The pyrealsense2 package should be in the system but either python3 or python can't load the package. I feel very frustrated now. 
My question: Is there anything ignored after finishing your script?
Thanks a lot for your help!
Best Regards,

GaoQi-2023, 2023-10-02:
Supplement: After finishing the bash script, I just tried "python3 _directory_/myscript.py" or "python _directory_/myscript.py". Does it need to cd to somewhere?

35selim, 2023-10-02:
Hi @GaoQi-2023,

To my knowledge, JetPack SDK 4.6 includes Ubuntu 18.04, which should not present compatibility issues with the RealSense SDK due to an ARM-based Ubuntu version.

The potential issue could be related to the Python version or Python path being used. If the Python version is greater than 3.6, consider downgrading to a compatible version. 
Additionally, verify the Python interpreter path, ensuring it matches the path used in the script: 
**-DPYTHON_EXECUTABLE=/usr/bin/python3**. Ensure it's python3.x, not python2. 
You can confirm the version and path of your Python interpreter with the following command: `which python3 && python3 -V`.

Best regards,

Selim

GaoQi-2023, 2023-10-03:
@35selim Thanks a lot! 

The default Python3 in the latest JetPack SDK 4.6 is 3.6.9, I didn't change it. 
I will try your suggestion later. 

Best regards,

Gao

MartyG-RealSense, 2023-10-17:
Hi @GaoQi-2023  Do you have an update about your problem that you can provide to @35selim please?  Thanks!

GaoQi-2023, 2023-10-17:
@MartyG-RealSense Hi, thanks for your reply. Sorry for not updating in time, since I was also working on other tasks. 
I have tried the command "which python3 && python3 -V", but the python version and interpreter path should be correct. Now I am still trying to solve the problem.

35selim, 2023-10-18:
Hi @GaoQi-2023 again,

Can you confirm that the requirements stated [here](https://github.com/IntelRealSense/librealsense/issues/10891#issuecomment-1275874984) are met?
If the installation is complete without giving any errors, there should be either Python path inconsistency or a software specific problem. (on the Jetson side since I tried the script recently on a Jetson device and pyrealsense2 seems to function well)

Best,
Selim

GaoQi-2023, 2023-10-18:
@35selim Thanks again!
I believe the problem definitely will be solved because indeed there are success cases. But now I am busy in another city. I will update the issue next week.

GaoQi-2023, 2023-10-24:
@35selim Hi
Today, I reinstalled the latest Jetpack and followed your bash scripts without any additional operation, but it still doesn't work. 

Yes, I found 3 files with names of "pyrealsense2.cpython-36m-aarch64-linux-gnu.so", "pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.54" and "pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.54.2". So, the pyrealsense2 should be built successfully. 

But, when I try to run my code (with "python3"), the terminal outputs: 

> ModuleNotFoundError: No module named 'pyrealsense2'

And if I try to import pyrealsense2 in the terminal (with command `import pyrealsense2`), it said: 

> import-im6.q16: not authorized `pyrealsense2' @ error/constitute.c/WriteImage/1037.

Now, I am pretty sure the interpreter path is `/usr/bin/python3` and the python version is `Python 3.6.9`. So, I totally have no idea about it.

35selim, 2023-10-26:
Hi again,

The error message **import-im6.q16: not authorized** can be caused by security policies in the ImageMagick software suite. If that is the case you can solve this with modifying the file **/etc/ImageMagick-6/policy.xml**. You can change the line in the file that includes **policy domain="coder" rights="none" pattern="PYREALSENSE2"**  so that **rights** section is **rights="read|write"**. 

Since you see the shared object files, you successfully installed the package but Python most probably cannot find them. You can simply check this with writing **export PYTHONPATH=$PYTHONPATH:/path/to/object/files** in the terminal and trying again running your script. If this solves your problem you simply add this export line to the end of your **~/.bashrc** file.

Best,
Selim

GaoQi-2023, 2023-11-01:
@35selim Thank you! For some reasons, now I set aside this RealSense work. 
Thanks again for your effort!

GaoQi-2023, 2023-11-09:
@35selim @MartyG-RealSense Good morning! 

Inspired by the topic "Not able to use pyrealsense2. "No Module named pyrealsense2" #12377", my issue has been solved.

Actually, the CMake built pyrealsense2 package files are in the `/home/*username*/librealsense/build/release` folder, instead of `/usr/local/lib/python3.6/pyrealsense2` folder. So, I used `sudo mkdir /usr/local/lib/python3.6/pyrealsense2` to create the folder, and move all *.so files (5 in total) into it with `sudo cp` command. And finally my script worked without any error and warning. 

So, I think the problem location in the `build_pyrealsense2_and_SDK.sh` should be the two lines:
```
echo 'PYTHONPATH="/usr/local/lib:/usr/local/lib/python3.6/pyrealsense2:$PYTHONPATH"' >> ~/.bashrc
echo 'export PYTHONPATH' >> ~/.bashrc
```

By the way, I installed OpenCV 4.5.1 with the help of [](https://github.com/Qengineering/Install-OpenCV-Jetson-Nano/blob/main/OpenCV-4-5-1.sh) in my previous work. I do not know if it contributes to the issue. 

Now I am working with ZED camera, but I will use RealSense in further work.

Thank you both very much!

Best Regards,

Gao

MartyG-RealSense, 2023-11-09:
Thanks so much @GaoQi-2023 for sharing the details of your solution!

Pragadesh-45, 2023-12-06:
> @35selim @MartyG-RealSense Good morning!
> 
> Inspired by the topic "Not able to use pyrealsense2. "No Module named pyrealsense2" #12377", my issue has been solved.
> 
> Actually, the CMake built pyrealsense2 package files are in the `/home/*username*/librealsense/build/release` folder, instead of `/usr/local/lib/python3.6/pyrealsense2` folder. So, I used `sudo mkdir /usr/local/lib/python3.6/pyrealsense2` to create it, and cpoied all *.so files (5 in total) into it with `sudo cp` command. And finally my script worked without any error and warning.
> 
> So, I think the problem location in the `build_pyrealsense2_and_SDK.sh` should be the two lines:
> 
> ```
> echo 'PYTHONPATH="/usr/local/lib:/usr/local/lib/python3.6/pyrealsense2:$PYTHONPATH"' >> ~/.bashrc
> echo 'export PYTHONPATH' >> ~/.bashrc
> ```
> 
> By the way, I installed OpenCV 4.5.1 with the help of "https://github.com/Qengineering/Install-OpenCV-Jetson-Nano/blob/main/OpenCV-4-5-1.sh" in my previous work. I do not know if it contributes to the issue.
> 
> Now I am working with ZED camera, but I will use RealSense in further work.
> 
> Thank you both very much!
> 
> Best Regards,
> 
> Gao

Hi @GaoQi-2023 - manual copy didn't work for me. My script only works when I add system path `/home/*username*/librealsense/build/release` explicitly

GaoQi-2023, 2023-12-06:
Hi, @Pragadesh-45 
Did you run the lines in the `build_pyrealsense2_and_SDK.sh` below?

> echo 'PYTHONPATH="/usr/local/lib:/usr/local/lib/python3.6/pyrealsense2:$PYTHONPATH"' >> ~/.bashrc
> echo 'export PYTHONPATH' >> ~/.bashrc

I know they led to the problem that troubled me for a long time. But it set `/usr/local/lib/python3.6/pyrealsense2` as a system path and made the path effective globally in the system. So, my manual copy could work. 

The two lines should be modified. You can do it as you wish. But I won't do it so far, because I have made my pyrealsense2 work and I don't want to be involved in another potential trouble.

closed 2026-10-23 by MartyG-RealSense