realsenseai/librealsense #10894: How to construct a pointcloud with only some points

alessio8russo, 2022-09-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400i | 
| Firmware Version                   |05.13.00.50| 
| Operating System & Version |   Win 10   |                                         | 
| Platform                                 | PC |
| Language                            |  python                     | 
| Segment			| others                          | 

### Issue Description

Hi, i'm very new with python and with  pyrealsense, so probably is an easy question. I construct a background removal+color mask in order to isolate only the cable , so all the pixel that are not of the cable are black, but obviously doing in this way when i construct the poincloud using open3d it give as result also the black point. This is my code:
```
from email.mime import image
import string
from turtle import shape
import pyrealsense2 as rs
import numpy as np
import cv2
import argparse
import time
import open3d
import os
import matplotlib.pyplot as plt
from skimage.filters import frangi

def rescaleFrame(frame,scale):
    width = int(frame.shape[1]*scale)
    height = int(frame.shape[0]*scale)
    dimensions = (width,height)
    return cv2.resize(frame, dimensions,interpolation=cv2.INTER_AREA)

if __name__=="__main__":
    time_start = time.time()
    var = True
    parser = argparse.ArgumentParser(description="Read recorded bag file and display depth stream in jet colormap.\
                                    Remember to change the stream resolution, fps and format to match the recorded.")
    # Add argument which takes path to a bag file as an input
    parser.add_argument("-i", "--input", type=str, help="Path to the bag file")
    # Parse the command line arguments to an object
    args = parser.parse_args()
    # Safety if no parameter have been given
    if not args.input:
        print("No input paramater have been given.")
        print("For help type --help")
        exit()
    # Check if the given file have bag extension
    if os.path.splitext(args.input)[1] != ".bag":
        print("The given file is not of correct file format.")
        print("Only .bag files are accepted")
        exit()

    hole_filling0 = rs.hole_filling_filter(0)
    align = rs.align(rs.stream.color)
    pipeline = rs.pipeline()
    config = rs.config()
    # Tell config that we will use a recorded device from filem to be used by the pipeline through playback.
    rs.config.enable_device_from_file(config, args.input)
    # Configure the pipeline to stream the depth stream
[... 139 lines cut ...]
            # if pressed escape exit program or q
            if key & 0xFF == ord('q') or key == 27 :
                cv2.destroyAllWindows()
                break

    finally:
        pipeline.stop()
        pass

```
So and I obtained a pointcloud like: 
[image]

So in order to resolve this problem i thought to use np.nonzero on the mask in order to have the index o pixel that are different from zero and then use this index in rs2_deproject_pixel_to_point(), but i have difficulties to define the depth value for each pixel , this is my code to check if they have the same depth that gives me error "IndexError: index 2 is out of bounds for axis 1 with size 2"
```
 index = np.transpose(np.nonzero(auxiliary_mask))
index_row = list(range(0, index.shape[0]))
index_col = list(range(0, index.shape[1]))
 for mm in index_row:
                for nn in index_col:
                    z = depth_frame0.get_distance(index[mm,mm],index[mm,nn])
                print(z)

```
And also in case i will able to use  rs2_deproject_pixel_to_point() , i didn't get how to use these points in order to construct and visualize the poincloud with open3d. So there is an easiest way to construct only the poincloud ?

MartyG-RealSense, 2022-09-13:
Hi @alessio8russo  May I first ask whether the cable is black?  Because of general physics principles (not RealSense specifically), color shades such as dark grey and black absorb light, making it difficult for depth cameras to analyze those surfaces for depth information.  So a depth image containing a black cable may appear to show a cable but it may actually just be a cable-shaped black empty area with no depth values in it.

If the cable is black, changing to a different colored cable would help it to be better detected.  Alternatively, projecting a strong light source onto a black area can help to bring out depth information from it.

alessio8russo, 2022-09-13:
Hi, thanks for reply so quickly, but the problem i think is not the color of cable, because i have a problem how to code what i need to do, because after applying the hole filling (  to resolve the problem of random hole wiht 0 depth  in the cable) i have resolved any problem with the depth and depth value. 
So my problem is how to have the poincloud with only the cable using the data from color masking that isolate the cable.
Because i tried also to do a depth filtering and obviously in this way the pointcloud isolate only the cable but all the processing on the color image in this case is useless.

MartyG-RealSense, 2022-09-13:
I note that you are applying the hole-filling post processing filter **before** the depth-color alignment is applied, which is the order that Intel recommends.  So the alignment part of the script is likely okay.

I also see that you seem to be taking account of the OpenCV default color space being BGR instead of RGB so that you are not getting a reverse-color image after converting the RealSense image to OpenCV.

The impression that I get from the image though is that the RGB image of the cable is not aligned with the depth image of the cable. 

[image]

Does the pointcloud look better if you test the Open3D pyrealsense2 pointcloud script at https://github.com/isl-org/Open3D/issues/473#issuecomment-508139946

alessio8russo, 2022-09-13:
this code is almost identical to mine , the only difference is the fact that i rescaled the depth and the color image, and without rescaling i have a better pointcloud, but the issue for which i have created this post  remain, i want only the cable without the black part. Where as  i said, i had also the black part because with the construction of poincloud it project also the part that i don't need of the masked color image (the res image below). For this I asked for suggestion how to have this with np.nonzero and rs2_deproject_pixel_to_point(), or if there are other way to obtain only the cable after i isolated it in the color image.
[image]

MartyG-RealSense, 2022-09-13:
The C++ **rs-grabcuts** OpenCV example that also creates a cut-out takes the approach of creating a foreground and background mask and then extracts foreground pixels.  Although it is a C++ example, its cv code may provide some useful insights about your own isolation project.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/grabcuts/rs-grabcuts.cpp#L87

alessio8russo, 2022-09-13:
Ok thank you for the reply, i will try to see, so the way of use  np.nonzero and rs2_deproject_pixel_to_point() isn't good? Because i have some doubts how to use it, so how to visualize the point that i obtain from rs2_deproject_pixel_to_point() ?

alessio8russo, 2022-09-13:
I have obtained what i need how we can see from the image: 
[image]

this is my code, if anyone have my same issues : 
```
from email.mime import image
import string
from turtle import shape
import pyrealsense2 as rs
import numpy as np
import cv2
import argparse
import time
import open3d
import os
import matplotlib.pyplot as plt
from skimage.filters import frangi

def rescaleFrame(frame,scale):
    width = int(frame.shape[1]*scale)
    height = int(frame.shape[0]*scale)
    dimensions = (width,height)
    return cv2.resize(frame, dimensions,interpolation=cv2.INTER_AREA)

if __name__=="__main__":
    time_start = time.time()
    var = True
    parser = argparse.ArgumentParser(description="Read recorded bag file and display depth stream in jet colormap.\
                                    Remember to change the stream resolution, fps and format to match the recorded.")
    # Add argument which takes path to a bag file as an input
    parser.add_argument("-i", "--input", type=str, help="Path to the bag file")
    # Parse the command line arguments to an object
    args = parser.parse_args()
    # Safety if no parameter have been given
    if not args.input:
        print("No input paramater have been given.")
        print("For help type --help")
        exit()
    # Check if the given file have bag extension
    if os.path.splitext(args.input)[1] != ".bag":
        print("The given file is not of correct file format.")
        print("Only .bag files are accepted")
        exit()

    hole_filling0 = rs.hole_filling_filter(0)
    align = rs.align(rs.stream.color)
    pipeline = rs.pipeline()
    config = rs.config()
    # Tell config that we will use a recorded device from filem to be used by the pipeline through playback.
    rs.config.enable_device_from_file(config, args.input)
    # Configure the pipeline to stream the depth stream
    config.enable_stream(rs.stream.depth,rs.format.z16, 30)
    config.enable_stream(rs.stream.color,rs.format.rgb8, 30)

    # Start streaming from file
    profile = pipeline.start(config)

    intr = profile.get_stream(rs.stream.color).as_video_stream_profile().get_intrinsics()
   
    pinhole_camera_intrinsic = open3d.camera.PinholeCameraIntrinsic(intr.width, intr.height, intr.fx, intr.fy, intr.ppx, intr.ppy)

    geometrie_added = False
    vis = open3d.visualization.Visualizer()

    vis.create_window("Pointcloud")
    pointcloud = open3d.geometry.PointCloud()
# Getting the depth sensor's depth scale (see rs-align example for explanation)
    depth_sensor = profile.get_device().first_depth_sensor()
    depth_scale = depth_sensor.get_depth_scale()
# # We will be removing the background of objects more than clipping_distance_in_meters meters away
[... 106 lines cut ...]

            if not geometrie_added:
                vis.add_geometry(pointcloud)
                geometrie_added = True
            #informs the vis that the related geometries are updated
            vis.update_geometry(pointcloud)
            #the visualizer renders a new frame by calling:
            vis.poll_events()
            vis.update_renderer()

            key = cv2.waitKey(1)

            

            # if pressed escape exit program or q
            if key & 0xFF == ord('q') or key == 27 :
                cv2.destroyAllWindows()
                break

    finally:
        pipeline.stop()
        pass
```
so i obtained only the cable (i don't know why it is blue), but there are already some outlier, any suggestion how to remove them? 
In addition to that there is a way in order to downsample the index of points that i have obtained using np.nonzero  on the mask to be less, so in this way the  pointcloud has lower number of points but only the most important? Because i need  the pointcloud to do a tracking of the cable, so have only the most important point is better.

alessio8russo, 2022-09-14:
Thank you so much. I will try with both. Is there any down-sampling on the color image or 1 channel-image?

MartyG-RealSense, 2022-09-14:
The 'High Accuracy' preset will only affect depth coordinates.  It does not down-sample, it just decides whether a depth point is included on the image or not based on how confident the SDK is in the accuracy of that point's depth value.

alessio8russo, 2022-09-14:
I read it, for this i ask for a method that can be applied on the color point , and so i can down-sample the color image,

alessio8russo, 2022-09-15:
Thank you, now i'm trying with outlier removal of opend3d, the problem is that if i add to my code:
pointcloud.voxel_down_sample(voxel_size=0.02) it hasn't any effect, then if i use
pointcloud = pointcloud.voxel_down_sample(voxel_size=0.02) , it has an effect but the pointcloud is not updated, i'm visualizing a single frame.
Any suggetion?

alessio8russo, 2022-09-15:
Thank you, now i'm trying with outlier removal of open3d that reduce also the size of point in the pointcloud

MartyG-RealSense, 2022-09-21:
Hi @alessio8russo  Do you require further assistance with this case, please?  Thanks!

alessio8russo, 2022-09-21:
No now i'm searching for other issues, thank you

MartyG-RealSense, 2022-09-21:
I researched Open3D interpolation in regards to RealSense.  Of the small number of references that were available, they were all interpolating with approximately the same **cv2.INTER_AREA** instruction that you are already using in your own script.

An example of such a reference is https://github.com/isl-org/Open3D/issues/3452

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a long time period.

closed 2026-10-23 by MartyG-RealSense