realsenseai/librealsense #10896: Unable to install pyrealsense2

eduardoafd, 2022-09-13:
I recently received my RealSense D405 and when trying to run the python examples I ran into the trouble of not having the pyrealsense2 installed. I than attempted runing the following command for installation:

`pip install pyrealsense2`

That resulted in the following error:

ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2

I'm currently running it on my PC with the following specs, any idea on what could fix this?
Windows 10
Intel Core i7 11800H

35selim, 2022-09-13:
Maybe you are using wrong python version as stated [#5777 (comment)](https://github.com/IntelRealSense/librealsense/issues/5777#issuecomment-582480988)

MartyG-RealSense, 2022-09-13:
Thanks for your advice @35selim, I was about to post about the Python version  :)

Hi @eduardoafd  Which Python version are you using, please?  The pip install method of installation supports installation o the pyrealsense2 wrapper from packages for Python 2.7, and Python 3.5 to 3.9.

If you are using a newer version such as Python 10 that pip install does not currently support then you can install the pyrealsense2 wrapper from source code.

eduardoafd, 2022-09-13:
Hi guys, a few moments after the issue was posted I came across the solution, unfortunately i could only come by to close the issue right now. The problem was that i was running python 3.10, **downgrading to python 3.7 solved the issue.**

Thanks @MartyG-RealSense and @35selim !

closed 2022-09-13 by eduardoafd