realsenseai/librealsense #10897: No computed baseline value set in /camera_info topic in the recorded rosbag file for right infrared sensor

vanguard478, 2022-09-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 (D435i) | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.4.0-121                                         | 
| Platform                                 | PC   |
|librealsense|2.51.1 RELEASE|
| Language                            | C++                          | 
| Segment			|  Others                      | 

### Issue Description
Hi, 
I am trying to record a ros bag file of both Left and Right Infrared images using the rs-record utility. So to enable the infrared streams I modified the `rs2::config cfg`  as follows to suit my needs:
 ```cpp
    cfg.enable_stream(RS2_STREAM_INFRARED, 1,848,480, RS2_FORMAT_Y8, 15);
    cfg.enable_stream(RS2_STREAM_INFRARED,2,848,480, RS2_FORMAT_Y8, 15);
    cfg.enable_record_to_file(out_file.getValue());
```
 As mentioned in the following [issue](https://github.com/IntelRealSense/realsense-ros/issues/1310) in the [realsense-ros repo](https://github.com/IntelRealSense/realsense-ros), the Projection matrix should have the P[0,3] = -fx * B in the Camera Info topic (according to [sensor_msgs/CameraInfo](http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/CameraInfo.html) )
 I recorded the bag file using the modified rs-record and there is no corresponding value for the computed -baseline*fx value in the Right Infrared Sensor Camera Info topic. The values are same for both left and right infrared camera topics. I even cross-checked with the realsense-viewer to record a rosbag file, and it also has the same issue for the Camera Info topic.
Although, the baseline value can be queried in the `´/device_0/sensor_0/option/Stereo_Baseline/value´` topic, it makes the bag file recorded by rs-record difficult to use with some SLAM algorithms which depend on the Camera_Info topic to query the baseline.

Is there any parameter in the `rs::config` I can change so that the camera_info topic value will be correctly set in stereo recording setup?

MartyG-RealSense, 2022-09-13:
Hi @vanguard478  I believe that this issue has been discussed on the RealSense ROS forum over several different cases - including https://github.com/IntelRealSense/realsense-ros/issues/1310 that you quoted - and so it can be a difficult subject to follow due to one case leading to another and then another.

It may therefore be best to link to a pull request at https://github.com/IntelRealSense/realsense-ros/pull/1242 that was eventually created and merged into the RealSense ROS wrapper at the end of the lengthy discussions.  

The comments in this discussion remark that infra2 used the frame_id of infra1.  This is because having the right camera frame_id equal to that of the left camera is required by `StereoCameraModel` from image_geometry and stereo_image_proc

This subject has been further discussed at https://github.com/IntelRealSense/realsense-ros/issues/2220#issuecomment-1010720769 where suggestions for dealing with it were offered.

 The information applies to the RealSense ROS wrapper though.  I am not aware of a method by which the issue could be addressed through librealsense SDK scripting, unfortunately.

MartyG-RealSense, 2022-09-19:
Hi @vanguard478  Do you require further assistance with this case, please?  Thanks!

vanguard478, 2022-09-21:
Hi, 
I referred to the issues which you pointed out and I think Camera Info matrix are properly set by the ROS wrapper. The issue with rs-record is still there. As of now I am just modifying the published message as per my requirement.

MartyG-RealSense, 2022-09-27:
Hi @vanguard478  Can this case be closed now or do you have further questions, please?  Thanks!

vanguard478, 2022-09-29:
Hi @MartyG-RealSense, I don't have any further question regarding this. I just hope this issue with the Camera Parameters being incorrectly set in the bag file recorded by the rs-record utility is fixed in an future release.

closed 2022-09-29 by vanguard478