realsenseai/librealsense #10900: Can I get disparity value from realsense D455

JasonChanJY, 2022-09-14:
I understand how to get depth value from D455，however，Is there any method to get disparity value directly？

MartyG-RealSense, 2022-09-14:
Hi @JasonChanJY  The information at https://github.com/IntelRealSense/librealsense/issues/10381 may be helpful to you for obtaining the disparity value with scripting.

JasonChanJY, 2022-09-15:
> Hi @JasonChanJY The information at #10381 may be helpful to you for obtaining the disparity value with scripting.

Thanks, I have one more question, what do "depth to disparity" and "disparity to depth" mean in RealSense Viewer? Can it transform depth to disparity or in verse?

MartyG-RealSense, 2022-09-15:
You are correct, they perform transformation between depth and disparity domains.  As indicated by the order of the filters listed in the Viewer's post-processing filter section and in the documentation link below, the Disparity to Depth filter should be applied **after** Depth to Disparity.  These two filters are applicable only to RealSense 400 Series stereo camera models such as the D455.

https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#using-filters-in-application-code

closed 2022-09-29 by MartyG-RealSense