realsenseai/librealsense #10901: How can I load d435i in unity, ubuntu?

Deok-min, 2022-09-14:
Hi, I'm trying to run the Unity Wrapper in ubuntu 18.05.
But I can't find how to build it. How can i build unity wrapper?
Thank you.

MartyG-RealSense, 2022-09-14:
Hi @Deok-min  At the bottom of the information listing for each RealSense SDK version on the **Releases** page at the link below, there is an 'Assets' file list.

https://github.com/IntelRealSense/librealsense/releases

In that file list is a Unity package file called **Intel. RealSense.unitypackage**

[image]

You can download this file to your computer and then launch Unity and go to the menu option **Assets > ImportPackage > Custom Package** and navigate to the location on your computer that you saved the package to.  Select the file and the package should import the RealSense Unity wrapper files into your project.

Deok-min, 2022-09-14:
it works, thank you!!

closed 2022-09-14 by Deok-min

MartyG-RealSense, 2022-09-14:
That's excellent to hear, @Deok-min - thanks very much for the update!