realsenseai/librealsense #10902: control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

amrapalikhandare2, 2022-09-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version | Ubuntu 20  | 
| Kernel Version (Linux Only)    | 5.10.104                                         | 
| Platform                                 | NVIDIA Jetson Nx |
| SDK Version                            | 2.51.1                         | 
| Language                            |  Cpp                          | 
| Segment			|  Robot                        | 
| realsense_ros version			|  v4.0.4                      | 

### Issue Description
After launching by command  
```ros2 launch realsense2_camera rs_launch.py```
I keep getting the error
``` 
[realsense2_camera_node-1]  14/09 17:31:20,936 WARNING [281472476891552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  14/09 17:31:21,089 WARNING [281472476891552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  14/09 17:31:21,342 WARNING [281472476891552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
...
```

Following is the full output that displays after launching the camera

```
[INFO] [launch]: All log files can be found below /home/rightbot/.ros/log/2022-09-14-17-31-12-940522-ubuntu-47043
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [47046]
[realsense2_camera_node-1] [INFO] [1663156874.616471375] [camera.camera]: RealSense ROS v4.0.4
[realsense2_camera_node-1] [INFO] [1663156874.616810512] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1663156874.616917425] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1663156875.111891766] [camera.camera]: Device with serial number 018322071462 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1663156875.114755875] [camera.camera]: Device with physical ID 2-3.3-3 was found.
[realsense2_camera_node-1] [INFO] [1663156875.114876803] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1663156875.116949036] [camera.camera]: Device with port number 2-3.3 was found.
[realsense2_camera_node-1] [INFO] [1663156875.117116493] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1663156875.140857334] [camera.camera]: getParameters...
[realsense2_camera_node-1] [WARN] [1663156875.146055885] [camera.camera]: JSON file provided doesn't exist! (/application/rightbot_ws/src/realsense/realsense-ros/realsense2_camera/config/high_accuracy.json)
[realsense2_camera_node-1] [INFO] [1663156875.146225677] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1663156875.146318062] [camera.camera]: Device Serial No: 018322071462
[realsense2_camera_node-1] [INFO] [1663156875.146394926] [camera.camera]: Device physical port: 2-3.3-3
[realsense2_camera_node-1] [INFO] [1663156875.146482798] [camera.camera]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1663156875.146619311] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1663156875.148091926] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1663156878.791794606] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1663156880.052438754] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1663156880.114970453] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1663156880.249313989] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1]  14/09 17:31:20,469 WARNING [281472476891552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1663156880.522522681] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1663156880.525715431] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1663156880.534353773] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1663156880.648117347] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1663156880.649612841] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1663156880.663337798] [camera.camera]: 
[realsense2_camera_node-1]  14/09 17:31:20,936 WARNING [281472476891552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  14/09 17:31:21,089 WARNING [281472476891552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  14/09 17:31:21,342 WARNING [281472476891552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

```

MartyG-RealSense, 2022-09-14:
Hi @amrapalikhandare2  Does the number of **control_transfer returned** warnings reduce if you add **initial_reset:=true** to your launch instruction to reset the camera at launch, please?

**ros2 launch realsense2_camera rs_launch.py initial_reset:=true**

amrapalikhandare2, 2022-09-15:
Hi @MartyG-RealSense 

They reduced a little but it was not a significant reduction.

amrapalikhandare2, 2022-09-15:
I received the warning once if I launch with ros2 run.
Following is the output of ros2 run:

```
amrapali@ubuntu:/application/bot_ws$ ros2 run realsense2_camera realsense2_camera_node --ros-args -p initial_reset:=true
[INFO] [1663235773.277724911] [camera]: RealSense ROS v4.0.4
[INFO] [1663235773.278025713] [camera]: Built with LibRealSense v2.51.1
[INFO] [1663235773.278157618] [camera]: Running with LibRealSense v2.51.1
[INFO] [1663235773.770198531] [camera]: Device with serial number 018322071462 was found.

[INFO] [1663235773.770472805] [camera]: Device with physical ID 2-3.3-4 was found.
[INFO] [1663235773.770569381] [camera]: Device with name Intel RealSense D435I was found.
[INFO] [1663235773.772135566] [camera]: Device with port number 2-3.3 was found.
[INFO] [1663235773.772271855] [camera]: Device USB type: 3.2
[INFO] [1663235773.772369711] [camera]: Resetting device...
 15/09 15:26:13,827 WARNING [281472913082784] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b

```

MartyG-RealSense, 2022-09-15:
If the warning is only received a small number of times during launch and then stops, or only appears once or twice per minute after launch, then it can be safely ignored.

amrapalikhandare2, 2022-09-15:
I get 15-20 warnings just after launch and then 6-7 warnings at a interval of 1-1.5 minutes continuously.

MartyG-RealSense, 2022-09-15:
6-7 warnings after launch every 1 to 1.5 minutes likely falls into the 'safe' category.  Ultimately, if performance does not seem to be degraded despite the warnings being generated then I would recommend ignoring them.

amrapalikhandare2, 2022-09-15:
Performance degrades after some time and then camera is disconnected.

MartyG-RealSense, 2022-09-15:
The **Resource temporarily unavailable** type of control_transfer returned error indicates a communication problem between the camera and computer.

Is the camera connected directly to the Jetson board or attached to a USB hub?

amrapalikhandare2, 2022-09-16:
The camera is connected directly to the Jetson board.

MartyG-RealSense, 2022-09-16:
Degradation of performance over time and then disconnection could be a symptom of a **memory leak**.  This is where the computer's available memory is consumed over time until a program becomes unstable.  

Would it be possible to use an Ubuntu system monitoring tool such as **htop** to monitor the amount of free memory over time and see whether the memory is reducing whilst the roslaunch is active, please?

MartyG-RealSense, 2022-09-22:
Hi @amrapalikhandare2  Do you require further assistance with this case, please?  Thanks!

amrapalikhandare2, 2022-09-27:
No, this issue does not occur on Jetpack 4.6 which has ubuntu kernel 4.9, so I am using that. Thanks.

MartyG-RealSense, 2022-09-27:
You are very welcome.  As you do not require further assistance, I will close this case.  Thanks for the update!

closed 2022-09-27 by MartyG-RealSense