realsenseai/librealsense #10904: All the 3D coordinates are zeros when using CUDA to transform depth maps to point clouds.

Shannon-Law, 2022-09-15:
In my C++ program, I transform depth maps to point clouds with CUDA when I use the codes from https://github.com/IntelRealSense/librealsense/blob/d5e775b75f7adbf75c911b7c85a4afd2f16defcb/src/cuda/cuda-pointcloud.cu. But, all the 3D coordinates of the point clouds are zeros. Why is that and how to solve it?

MartyG-RealSense, 2022-09-15:
Hi @Shannon-Law  There should not be a need to use code from the quoted script to generate a CUDA-enabled point cloud.  If CUDA support is enabled in librealsense then CUDA acceleration should be applied to the pointcloud automatically when it is generated in librealsense or in the RealSense ROS wrapper.

CUDA support will be included if librealsense is installed from Debian packages, or if it is built from source code with CMake with the build flag **DBUILD_WITH_CUDA=true** included in the CMake build instruction.  

CUDA support is not included if librealsense was built from packages using the 'Method 1' instructions of the RealSense ROS wrapper though.

Shannon-Law, 2022-09-15:
Hi, @MartyG-RealSense, I used "apt-get" to install RealSense SDK according to https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md. Because of the fact that the SDK does not support the function rscuda::deproject_depth_cuda, I added the source codes above to my program to generate point clouds.

Shannon-Law, 2022-09-15:
By the way, when I use the codes from https://github.com/IntelRealSense/librealsense/blob/d5e775b75f7adbf75c911b7c85a4afd2f16defcb/src/cuda/cuda-pointcloud.cu, I add the function printf() in the kernel function. And printf() does not execute instead.

MartyG-RealSense, 2022-09-15:
CUDA support in the SDK provides acceleration for pointclouds, depth-color alignment and color conversion and does not accelerate any other type of operation.

If you are using the rs2::pointcloud instruction like you do in https://github.com/IntelRealSense/librealsense/issues/10672#issuecomment-1182668618 then the pointcloud generation should be receiving the benefit of CUDA acceleration automatically.

If you are using Jetson then the distribution_linux.md instructions are not suited for Jetson.  Instead, the package installation procedure at the link below should be used to install librealsense.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

Shannon-Law, 2022-09-15:
Now I run the program in my laptop.

MartyG-RealSense, 2022-09-15:
CUDA support requires an Nvidia GPU and will not work with any other brand of GPU.  Jetsons have an Nvidia GPU built into them.

If your laptop does not have an Nvidia GPU then the SDK provides an alternative acceleration system called **GLSL**.  You can enable GLSL support in your program by changing **rs2::pointcloud** references to **rs2::gl::pointcloud**

Shannon-Law, 2022-09-15:
I have an NVIDIA GPU in my laptop.

MartyG-RealSense, 2022-09-15:
Thank you for the confirmation that you have an Nvidia GPU.

Have you confirmed that CUDA acceleration is working on your laptop by trying another type of CUDA acceleration operation supported in the SDK such as depth-color alignment?

Shannon-Law, 2022-09-15:
No, I have not. I just need to generate point clouds with CUDA acceleration.

MartyG-RealSense, 2022-09-15:
The SDK's CUDA acceleration is almost always used with Jetson boards rather than PCs with an Nvidia GPU (built-in or video card).  So it is important to determine whether the SDK's CUDA support is working on your Nvidia-equipped laptop.  

https://github.com/IntelRealSense/librealsense/issues/9619 is an example of a laptop with an Nvidia GPU that had problems with CUDA.   The RealSense user in that case posted the solution for their particular situation at https://github.com/IntelRealSense/librealsense/issues/9619#issuecomment-898413093

Shannon-Law, 2022-09-15:
When I run the program in my Jetson board, the same problem remains.

MartyG-RealSense, 2022-09-15:
Can you tell me more please about how your pointcloud program works and why you need to change the cuda-pointcloud.cu code.  

It sounds as though you want to use deprojection to generate a 3D pointcloud and then get the XYZ values of the pointcloud.  You also want to use CUDA to accelerate that pointcloud.  Is that correct, please?

Given that your previous case at https://github.com/IntelRealSense/librealsense/issues/10672 also had a problem with zero values from the pointcloud, perhaps this case is a continuation of the same problem?

Shannon-Law, 2022-09-15:
Yes, you are right. I add cuda-pointcloud.cu to my C++ project to generate 3D point clouds in a fast way. In this project, the cu file is the only CUDA source file.

This problem is different from the previous case at https://github.com/IntelRealSense/librealsense/issues/10672, due to the fact that the functions of generating point clouds are different though.

MartyG-RealSense, 2022-09-15:
The only librealsense case that I know of related to **cuda-pointcloud.cu** and obtaining pointcloud coordinates is https://github.com/IntelRealSense/librealsense/issues/6443#issuecomment-635321693

Shannon-Law, 2022-09-18:
When I deleted the statement "set(CUDA_NVCC_FLAGS ${CUDA_NVCC_FLAGS} -gencode arch=compute_60,code=sm_60)" in the file CMakeLists.txt of my program, all the results were normal with CUDA.

MartyG-RealSense, 2022-09-18:
That's excellent news @Shannon-Law that you achieved a solution.  Thanks very much for the update  :)

Shannon-Law, 2022-09-18:
@MartyG-RealSense You're welcome!

closed 2022-09-24 by MartyG-RealSense