realsenseai/librealsense #10907: Convert point cloud to numpy array

kankanzheli, 2022-09-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   |  | 
| Operating System & Version |   Ubuntu18  | 
| Kernel Version (Linux Only)    |  5.4.0-125-generic  | 
| Platform                                 | PC// NVIDIA 3070  |
| SDK Version                            |  2.51.1  | 
| Language                            |  python  | 
| Segment			|  Robot   | 

### Issue Description
 Use ros to start the "realsense" camera. The image resolution is "(1280,720)" Get the real-time point cloud from the topic/camera/depth/color/points. How can this point cloud information be converted into numpy array in real time?

When I work with point clouds obtained by other cameras, I use the following code to convert point clouds into numpy matrices:

pc = ros_ numpy. numpify(ptcloud_data) #pc.shape=(720,1280)
height = pc.shape[0]
width = pc.shape[1]
points_ np_ global1 = np. zeros((height * width, 3), dtype=np.float32)
points_ np_ global1[:, 0] = np. resize(pc['x'], height * width)
points_ np_ global1[:, 1] = np. resize(pc['y'], height * width)
points_ np_ global1[:, 2] = np. resize(pc['z'], height * width)
points_ np_ global = np.reshape(points_np_global1,(720,1280,3))

However, when trying to work on the point cloud of realness, pc. shape=(xxxx,). And the value of xxxx is uncertain.

kankanzheli, 2022-09-16:
I think I have obtained point clouds from the topic of /camera/depth/color/points. I successfully observed the point cloud results from rviz. However, the conversion to numpy array failed.

kankanzheli, 2022-09-16:
I used the camera from the wrapper. Use the directive roslaunch realsense 2_ camera rs_ camera. launch
And change the point cloud part of "rs_camera. launch" to true

I think I have obtained point clouds from the topic of /camera/depth/color/points. I successfully observed the point cloud results from rviz. However, the conversion to numpy array failed.

MartyG-RealSense, 2022-09-16:
I wonder whether the information in https://github.com/IntelRealSense/librealsense/issues/1466 is what you have in mind for storing pointcloud data in a numpy array.

kankanzheli, 2022-09-16:
I also have an answer to this question. But I didn't fully understand its meaning. Are color and depth RGB image data and depth data respectively? Is the point cloud here calculated from these two types of data?

MartyG-RealSense, 2022-09-16:
Depth = depth and color = RGB, yes.  A point cloud can be generated from depth data only (a cloud of 3D depth points), but you can also map color coordinates onto 3D depth coordinates to produce a **textured** pointcloud.

kankanzheli, 2022-09-16:
Do the RGB map and depth data here need to be processed as "null array"? Or is it OK to directly process the data received by ROS after receiving it from ROS?
What packages need to be imported when using these codes?

kankanzheli, 2022-09-16:
Do the RGB map and depth data here need to be processed as "numpy array"? Or is it OK to directly process the data received by ROS after receiving it from ROS?
What packages need to be imported when using these codes?

kankanzheli, 2022-09-16:
Do the RGB map and depth data here need to be processed as "null array"? Or is it OK to directly process the data after receiving it from ROS?
What packages need to be imported when using these codes?

MartyG-RealSense, 2022-09-17:
For a basic pointcloud you should be able to directly make use of the data without installing additional packages and make changes to the pointcloud using the parameters described in the image below.

[image]

For a more advanced pointcloud application such as SLAM navigation, additional packages may need to be installed.  An example of this is Intel's D435i SLAM tutorial at the link below.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

kankanzheli, 2022-09-17:
Sorry, I don't think you understand my question. You provide a method to obtain the numpy array of the point cloud（https://github.com/IntelRealSense/librealsense/issues/1466）. But there are only two lines of code in that website. I don't quite understand how those two lines of code work. So I asked the above question. I just need to know how to convert the point cloud received by ros into numpy array The reason why I ask questions here is that when other cameras (kinect and zed) work on ros, I can successfully convert their point cloud information into numpy array. But I try to make that code work on the realsence camera, it is failed.

MartyG-RealSense, 2022-09-17:
Converting RealSense data to numpy is a subject that has been attempted by other RealSense users but does not yet have a definitive solution as far as I am aware, unfortunately.  The difficulties are discussed in https://github.com/IntelRealSense/librealsense/issues/7531

kankanzheli, 2022-09-17:
Can you tell us the difference between the pointcloud2 data obtained by realness from ros and the pointcloud2 data obtained by other cameras from ros?

MartyG-RealSense, 2022-09-17:
A possible difference that comes to mind is that rs_camera publishes an **unordered** pointcloud by default, whilst the other camera types may be using an ordered cloud.  An ordered pointcloud can be published by adding **ordered_pc:=true** to the RealSense roslaunch instruction.

**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud ordered_pc:=true**

kankanzheli, 2022-09-17:
Thank you. It works.

MartyG-RealSense, 2022-09-17:
That's great to hear!  Thanks very much for your patience  :)

closed 2022-09-23 by MartyG-RealSense