realsenseai/librealsense #10908: Intel Camera not detecetd with boxer clearpath

CarlosBambinos, 2022-09-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2022-09-16:
Hi @CarlosBambinos  Are you using the Boxer Indoor Mobile Robot by Clearpath Robotics, please?

https://clearpathrobotics.com/boxer/

If you are, is it the **Boxer 2.4** model released in November 2021, described in the link below?

https://www.clearpathrobotics.com/assets/guides/noetic/boxer/index.html

MartyG-RealSense, 2022-09-23:
Hi @CarlosBambinos  Do you require further assistance with this case, please?  Thanks!

CarlosBambinos, 2022-09-23:
> Hi @CarlosBambinos Do you require further assistance with this case, please? Thanks!

No sorry I think I made a mistake opening this one because you already answered me on the other question.

Thanks

closed 2022-09-23 by CarlosBambinos

MartyG-RealSense, 2022-09-23:
No problem at all, @CarlosBambinos - thanks very much for the update!