realsenseai/librealsense #10909: Camera Not detected with USB3

CarlosBambinos, 2022-09-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455 | 
| Firmware Version                   | 05.13.00.50| 
| Operating System & Version |   Linux (Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-20-generic x86_64 (64 bit)   | 
| Platform                                 | Machine:   System: Default string product: Default string v: Default string  |
| SDK Version                            |                         | 
| Language                            |        English, French         | 
| Segment			| Robot (Boxer from Clearpath)  | 

### Issue Description
Hello,

I'm having trouble when connecting a Intel D455 Camera to the robot Boxer from Clearpath.

To explain it clearly,

I have a Intel D455 on the robot, when I'm connecting the camera to my PC (laptop) with the original cable it's detected as USB 2.1 in realsense-viewer but I can't receive frame.
But when using the same cable on the robot it's detected and working using ROS package (but not working with my script, having the same error of this issue : https://github.com/IntelRealSense/librealsense/issues/2818 )

But when using a USB3.2 cable, it's working on my laptop perfectly (the script is working to) but it's not even detected on the robot...

If you had a similar issue I'm interrested

MartyG-RealSense, 2022-09-16:
Hi @CarlosBambinos  The **Couldn't resolve requests** error occurs when a camera stream configuration is requested that is not supported by the camera at the time that the request is made.  

This could be due to a typing error in the stream configuration code in the script, but is more likely to be due to the camera being detected as being on a USB 2.1 connection instead of USB 3.  This is because the range of supported resolutions and FPS speeds is more limited on USB 2.1 compared to the full range of resolution and FPS modes supported on a USB 3 connection.

If you are using the RealSense ROS wrapper then the camera may be working with that because the wrapper will automatically use the current default stream settings of that particular camera model if a custom stream configuration has not been defined.

As the camera is detected on the robot when using the original cable but not working with your script, would it be possible to post your script in a comment so that I can check the stream configuration in the code?  Thanks!

CarlosBambinos, 2022-09-19:
Yes of course, here it is:

`
#!/usr/bin/env python3
from re import X
import cv2
import pyrealsense2 as rs
import numpy as np
from flask import Flask, render_template, Response, request
import json
import requests

# Server Initialization
app = Flask('hello')

#Initialization camera 

context = rs.context()
pipe = rs.pipeline(context)
config = rs.config()
pipeline_wrapper = rs.pipeline_wrapper(pipe)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))

found_rgb = False
for s in device.sensors:
    if s.get_info(rs.camera_info.name) == 'RGB Camera':
        found_rgb = True
        break
if not found_rgb:
    print("The demo requires Depth camera with Color sensor")
    exit(0)

width = 640
height = 480
framerate = 60

config.enable_stream(rs.stream.depth, width, height, rs.format.z16, framerate)
config.enable_stream(rs.stream.color, width, height, rs.format.bgr8, framerate)
profile = pipe.start(config)

sensor_dep = profile.get_device().first_depth_sensor()
sensor_dep.set_option(rs.option.enable_auto_exposure, 1)
temporal = rs.temporal_filter(0.4,20,7)
hole_filling = rs.hole_filling_filter(2)
colorizer = rs.colorizer()

#Centralize all camera flux
class ImgCapture():
    def __init__(self):
        pass
    # Return color_image (RGB flux), images combination of depth and color_image without background and aligned_depth_frame only used to get distance of point
    def read(self):
        
        # Get depth_scale
        depth_sensor = profile.get_device().first_depth_sensor()
        depth_scale = depth_sensor.get_depth_scale()

        # All image after clipping_distance_in_meters will be removed
        clipping_distance_in_meters = 2 #meter
        clipping_distance = clipping_distance_in_meters / depth_scale

        # align is used to aligned depth and color frame
        align_to = rs.stream.color
        align = rs.align(align_to)

        # Get frameset of color and depth
        frames = pipe.wait_for_frames()

[... 143 lines cut ...]
<body>
<div class="container">
    <div class="row">
        <div class="col-lg-8  offset-lg-2">
            <h3 class="mt-5">Live Streaming</h3>
            [image]
            [image]
        </div>
    </div>
</div>
</body>        
    """

if __name__=='__main__':

    # Start streaming
    capture0 = ImgCapture()
    capture_depth_stream = ImgDepthStream(capture0)
    capture_color_stream = ImgColorStream(capture0)
    capture_depth = ImgDepth(capture0)
    app.run(host="0.0.0.0")
`

To explain it faslty, it is supposed to stream depth/RGB over the network using Flask (and I'm using a Jetson to do some Computer Vision on this).
The reason I'm using this script is because we had troubles with depth streaming with ROS

MartyG-RealSense, 2022-09-19:
If the D455 camera was being detected as being on a USB 2.1 connection then 60 FPS would be unsupported, causing a **Couldn't resolve requests** error.  This is because 30 FPS is the highest speed available at 640x480 on USB 2.1, with 60 FPS being supported at 480x270 resolution.

[image]

[image]

CarlosBambinos, 2022-09-19:
> If the D455 camera was being detected as being on a USB 2.1 connection then 60 FPS would be unsupported, causing a **Couldn't resolve requests** error. This is because 30 FPS is the highest speed available at 640x480 on USB 2.1, with 60 FPS being supported at 480x270 resolution.
> 
> **DEPTH**
> 
> [image]
> 
> [image]
> 
> **RGB**
> 
> [image]

Yes you are right.

I put only 15 FPS and it's working. 

I tried 30 FPS but the stream is lagging, I don't know if it's because of my connection or something else but for the moment I will stay with 15 FPS.

I still don't know why USB3 cable doesn't work but at least I'm not blocked by this problem.

Thanks a lot !

closed 2022-09-19 by CarlosBambinos

MartyG-RealSense, 2022-09-19:
You are very welcome  :)