realsenseai/librealsense #10910: Point pose for robot interaction.

eduardo-inwound, 2022-09-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405| 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |2.51.1.4348| 
| Language                            |  Python                          | 
| Segment			|  Robot                          | 

First of all, I apologize in advance if this has already been covered in another issue I've searched but could not find this specific thing.

I have my D405 attached to the end-effector of a robotic arm and I've managed to obtain a point cloud of the surface i'm interacting with. My goal is to obtain a pose for the robot at each individual point of the point cloud as if it was moving with it's end-effector normal to the surface. I've already solved all of the robot's transformations so if I just need to find a way to define the point orientation relative to camera in Euler angles (Rx, Ry, Rz), is there a way to obtain that pose from the point cloud? Could anyone point me in the direction I should follow?

MartyG-RealSense, 2022-09-17:
Hi @eduardo-inwound  I would recommend https://github.com/IntelRealSense/librealsense/issues/5553 and https://github.com/IntelRealSense/librealsense/issues/7560 as starting points for research of this subject.

MartyG-RealSense, 2022-09-23:
Hi @eduardo-inwound  Do you require further assistance with this case, please?  Thanks!

eduardo-inwound, 2022-09-28:
Hi, I did have a look at those referencesbut still haven't managed to implement it, I'll update here once I have news

MartyG-RealSense, 2022-09-28:
Thanks very much @eduardo-inwound for the update - good luck!

eduardo-inwound, 2022-10-07:
Finally managed to take time to implement it, i'll explain in how I've done it in case anyone ever needs it:

1. Acquired point cloud.
2. After acquiring the point cloud all of the vertices were mapped from the camera reference frame to the robot's world reference frame. 
3. Let a point _p_ be the interest vertex to which the robot normal pose must be calculated with pixel coordinates _px, py_ and _a, b, c, d_ the adjacent points in the camera frame with known coordinates in the global coordinate system. Those points are arranged in such way:
```
       c
       |
a -- p -- b
       |
      d
```
4. To define the needed Euler Angles a new coordinate system must be defined, for that we need to define xp, yp and zp vectors which represent the three axes of the coordinate system in the point p.

`
        x_p = b - a
        x_p = x_p / np.linalg.norm(x_p)

        y_p = d - c
        y_p = y_p / np.linalg.norm(y_p)

        z_p = np.cross(y_p, x_p)
        x_p = np.cross(y_p, z_p)
`
5. Since the coordinate system must be by an orthonormal base of vectors in order to represent a rotation matrix and  x_dir and y_dir are not necessarily normal vectors  we perform the cross product of  y and z to obtain a new x which is normal to both z and y.
6. With those three vectors a rotation matrix can be calculated with the help of scipy
```
from scipy.spatial.transform import Rotation

r = Rotation.from_matrix([[x_p[0], y_p[0], z_p[0]],
                                         [x_p[1], y_p[1], z_p[1]],
                                         [x_p[2], y_p[2], z_p[2]]])
```

7. With the help of scipy once again we can calculate the rotation angles. It is important to say here that we are considering here XYZ fixed angle coordinate system for the rotation.
`
rx, ry, rz = r.as_euler('xyz', degrees=True)
`

Now you have 6D pose for your end-effector : )

closed 2022-10-07 by eduardo-inwound

MartyG-RealSense, 2022-10-08:
Thanks so much @eduardo-inwound for sharing your detailed feedback of your solution   :)