realsenseai/librealsense #10911: Latest Release (v2.51.1) Does NOT Have CUDA Enabled

luis-camero, 2022-09-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version | Ubuntu 20.04 | 
| Kernel Version (Linux Only)    | 5.4.0-125-generic | 
| Platform                                 | PC |
| SDK Version                            | 2.51.0 | 
| Language                            |  C++/Python (using `realsense-ros`)  | 
| Segment			|  Robot |                          | 

### Issue Description
With the previous release (v2.50.0), the `librealsense2` pre-built binaries would have been built with CUDA (i.e. `-DBUILD_WITH_CUDA=True`), and building the `realsense-ros` package with these pre-built binaries and CUDA would result in the Realsense ROS node using the GPU for filters (such as shading pointclouds).

This was confirmed by @dorodnic [here](https://github.com/IntelRealSense/librealsense/issues/6302#issuecomment-619531064).

However, it seems that the latest pre-built binaries of the **librealsense2 SDK (v2.51.1)** was not built with CUDA enabled. 

Previously (2.50.0-0-realsense), whenever the Realsense ROS was started, it would be registered in `nvtop`, clearly utilizing the GPU.
Now (2.51.1-0-realsense), does not. 

If instead, I manually build the latest librealsense SDK (v2.51.1) from source with the `-DBUILD_WITH_CUDA=True` compiler flag and build the `realsense-ros` against it, then the Realsense node will utilize the GPU.

**Was disabling the `BUILD_WITH_CUDA` compiler flag deliberate, or an oversight? Should we expect all upcoming pre-built packages to be built without CUDA enabled?**

MartyG-RealSense, 2022-09-17:
Hi @luis-camero  The Debian packages should have CUDA support enabled by default (though the packages installed by the RealSense ROS wrapper's 'Method 1' apt-get installation procedure do not have CUDA support included).

Let's keep this case open to see whether other RealSense users report the same issue with Debian package installations.  Thanks for the report!

MartyG-RealSense, 2022-09-20:
CUDA support is enabled in the librealsense SDK and does not need to be enabled in the ROS wrapper in order to automatically accelerate pointclouds and depth-color alignment generated by the wrapper.

I will consult with my Intel RealSense colleagues about your problem with SDK 2.51.1.  Thanks very much for your patience!

Nir-Az, 2022-09-21:
Hi @luis-camero ,

It could be that the configuration that was used to build until version 2.50 is different than the one we used for building 2.51.1 (Sorry for that).

I can confirm that release 2.51.1 was built with this configuration:
[JetPack 4.4.1 (L4T 32.4.4)](https://developer.nvidia.com/embedded/linux-tegra-r3244)
Which includes CUDA version 10.2 .

I would suggest to verify the installed CUDA version and try to match it, it should work since we do build it with `-DBUILD_WITH_CUDA=True`.

Good luck,
Nir

MartyG-RealSense, 2022-09-27:
Hi @luis-camero  Do you require further assistance with this case, please?  Thanks!

luis-camero, 2022-09-27:
Hi @Nir-Az,  
I found this issue when using setting up the ROS Realsense Noetic package using the 20.04 version of librealsense2-2.51.1 while using the latest stable release of CUDA (v11.7).

I cannot easily match CUDA v10.2 in Ubuntu 20.04, since v10.2 was not released for 20.04. 

However, I have a few questions:
1. I'm confused as to how the librealsense SDK was built for Ubuntu 20.04 in JetPack; I thought these two platforms had conflicting architectures
2. Where previous SDK versions built on JetPack as well?

Thanks for the help, 
Luis

Nir-Az, 2022-09-27:
Hi @luis-camero ,
1. The `LibRealSense` SDK was built using [JetPack 4.4.1](https://developer.nvidia.com/embedded/jetpack) which is based on Ubuntu 18.04 (not for Ubuntu 20.04)
2. Yes, the previous (arm64) SDK Debians were built on a Jetson platform, I am not sure if it was the exact same version.

Since we do not supply prebuilt packages for all JP versions you can always build from source on your platform and it should work (like we saw on your case).

Hope this information helps..
Nir

MartyG-RealSense, 2022-10-03:
Hi @luis-camero  Was the information provided in the comment above helpful, please?  Thanks!

luis-camero, 2022-10-11:
I understand that I have to build from source. But I am NOT using a Jetson. I am using a standard `Ubuntu 20.04 server amd64` install and pulled the compatible `librealsense2`  amd64, pre-built package from your package server. 

I am confused as to how or why you'd build `amd64` packages for Focal in Jetpack with 18.04. There must be a different build configuration and different CUDA version for this `2.51.1-0~realsense0.7526` package.

MartyG-RealSense, 2022-10-11:
Hi @luis-camero  If you are installing from packages on an amd64 computer then the instructions on the **distribution_linux.md** page should be used.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

JetPack is only involved in installations on Jetson boards, not amd64 computers.  @Nir-Az referred to **arm64** rather than amd64 in his advice above at https://github.com/IntelRealSense/librealsense/issues/10911#issuecomment-1259837645 - arm64 refers to computers / computing devices that use CPUs with Arm architecture, whilst amd64 usually refers to x86 / x64 CPU architecture like that in desktop and laptop PCs.

luis-camero, 2022-10-11:
I understand, thanks!

closed 2022-10-11 by luis-camero