realsenseai/librealsense #10914: The requirement of width of external trigger signal

ye-evercoast, 2022-09-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Kubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.11.0-25-generic   | 
| Platform                                 | PC  |
| SDK Version                            |  v2.51.1 | 
| Language                            |  C/C++                          | 
| Segment			|  VR/AR                          | 

### Issue Description
Hi, I'm trying to investigate some syncing issues in our multi camera system. We're using a master/slave camera topo, using master's trigger signal to sync all the slaves. The first thing I do is checking the master camera's trigger signals, as suggested in Intel Realsense's document: https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#e-external-trigger

I found the frequency of the signal quite close to the nominal one, while the signal width is only about 50ns, which is half of what suggested in the document mentioned above: `When applying an external sync pulse, the HW SYNC input requires a 100 microsecond positive pulse at the nominal camera frame rate, 33.33 ms for a 30Hz frame rate for example. Inputs are high impedance, 1.8V CMOS voltage levels.`

Attached my scope's image from the pin 5 when turned the "Output Trigger Enabled" on in Realsense Viewer in the "Stereo Module > Control" section.
[image]

I think my question is simple: is the 100ns signal width a hard requirement, as I see in the camera's self trigger signal is only 50ns? Could that be the root of some syncing issues?

MartyG-RealSense, 2022-09-19:
Hi @ye-evercoast  The 100 microsecond value applies to an 'external trigger' - one that is produced by a signal generator device instead of a RealSense camera set to master.  The master trigger from a camera is set to 50 microseconds, as confirmed by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/2890#issuecomment-449171468

ye-evercoast, 2022-09-19:
Thanks for the prompt reply. That's very insightful comment and answers my question.

closed 2022-09-19 by ye-evercoast