realsenseai/librealsense #10919: Linker error when building custom application with g++

mathklk, 2022-09-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |  Linux arm64 (Raspberry Pi 4) | 
| Kernel Version (Linux Only)    |  5.15.61-v8+                                         | 
| Platform                                 | Raspberry Pi 4 Model B Rev 1.4 |
| SDK Version                            |  2.44.0                          | 
| Language                            |  C++                        | 
| Segment			|  others                          | 

### Issue Description
I'm trying to use the rs2 library in my custom c++ application, which I compile using g++.  I successfully built the SDK from source and tested it with the hello-realsense example. I had to add the following to lines to the CMakeList.txt, but after that it compiled and worked fine.

```
find_package(realsense2 2.24.0)
target_link_libraries(rs-hello-realsense ${DEPENDENCIES} ${realsense2_LIBRARY})
```

The issue arises when I try to include it in my custom application, which does not use cmake, but g++. 
This is the g++ command I use:

```
/usr/bin/g++ -std=c++17 -o out -I"/usr/include/" -L/usr/local/lib/aarch64-linux-gnu/ -lrealsense2 -lpthread main.cpp
```

This produces the following error output:
```
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::error::error(rs2_error*)':
main.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x1c): undefined reference to `rs2_get_error_message'
/usr/bin/ld: main.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x58): undefined reference to `rs2_get_failed_function'
/usr/bin/ld: main.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x6c): undefined reference to `rs2_get_failed_function'
/usr/bin/ld: main.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0xcc): undefined reference to `rs2_get_failed_args'
/usr/bin/ld: main.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0xe0): undefined reference to `rs2_get_failed_args'
/usr/bin/ld: main.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x140): undefined reference to `rs2_get_librealsense_exception_type'
/usr/bin/ld: main.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x154): undefined reference to `rs2_free_error'
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::error::handle(rs2_error*)':
main.cpp:(.text._ZN3rs25error6handleEP9rs2_error[_ZN3rs25error6handleEP9rs2_error]+0x20): undefined reference to `rs2_get_librealsense_exception_type'
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::stream_profile::stream_profile(rs2_stream_profile const*)':
main.cpp:(.text._ZN3rs214stream_profileC2EPK18rs2_stream_profile[_ZN3rs214stream_profileC5EPK18rs2_stream_profile]+0xa0): undefined reference to `rs2_get_stream_profile_data'
/usr/bin/ld: main.cpp:(.text._ZN3rs214stream_profileC2EPK18rs2_stream_profile[_ZN3rs214stream_profileC5EPK18rs2_stream_profile]+0xb8): undefined reference to `rs2_is_stream_profile_default'
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::frame::~frame()':
main.cpp:(.text._ZN3rs25frameD2Ev[_ZN3rs25frameD5Ev]+0x24): undefined reference to `rs2_release_frame'
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::frame::get_profile() const':
main.cpp:(.text._ZNK3rs25frame11get_profileEv[_ZNK3rs25frame11get_profileEv]+0x24): undefined reference to `rs2_get_frame_stream_profile'
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::frame::add_ref() const':
main.cpp:(.text._ZNK3rs25frame7add_refEv[_ZNK3rs25frame7add_refEv]+0x1c): undefined reference to `rs2_frame_add_ref'
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::frame::reset()':
main.cpp:(.text._ZN3rs25frame5resetEv[_ZN3rs25frame5resetEv]+0x24): undefined reference to `rs2_release_frame'
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::video_frame::video_frame(rs2::frame const&)':
main.cpp:(.text._ZN3rs211video_frameC2ERKNS_5frameE[_ZN3rs211video_frameC5ERKNS_5frameE]+0x5c): undefined reference to `rs2_is_frame_extendable_to'
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::video_frame::get_width() const':
main.cpp:(.text._ZNK3rs211video_frame9get_widthEv[_ZNK3rs211video_frame9get_widthEv]+0x28): undefined reference to `rs2_get_frame_width'
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::video_frame::get_height() const':
main.cpp:(.text._ZNK3rs211video_frame10get_heightEv[_ZNK3rs211video_frame10get_heightEv]+0x28): undefined reference to `rs2_get_frame_height'
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::depth_frame::depth_frame(rs2::frame const&)':
main.cpp:(.text._ZN3rs211depth_frameC2ERKNS_5frameE[_ZN3rs211depth_frameC5ERKNS_5frameE]+0x5c): undefined reference to `rs2_is_frame_extendable_to'
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::depth_frame::get_distance(int, int) const':
main.cpp:(.text._ZNK3rs211depth_frame12get_distanceEii[_ZNK3rs211depth_frame12get_distanceEii]+0x38): undefined reference to `rs2_depth_frame_get_distance'
[... 20 lines cut ...]
main.cpp:(.text._ZN3rs26configC2Ev[_ZN3rs26configC5Ev]+0x20): undefined reference to `rs2_create_config'
/usr/bin/ld: main.cpp:(.text._ZN3rs26configC2Ev[_ZN3rs26configC5Ev]+0x2c): undefined reference to `rs2_delete_config'
/usr/bin/ld: main.cpp:(.text._ZN3rs26configC2Ev[_ZN3rs26configC5Ev]+0x30): undefined reference to `rs2_delete_config'
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::config::enable_stream(rs2_stream, int, int, int, rs2_format, int)':
main.cpp:(.text._ZN3rs26config13enable_streamE10rs2_streamiii10rs2_formati[_ZN3rs26config13enable_streamE10rs2_streamiii10rs2_formati]+0x58): undefined reference to `rs2_config_enable_stream'
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::config::can_resolve(std::shared_ptr<rs2_pipeline>) const':
main.cpp:(.text._ZNK3rs26config11can_resolveESt10shared_ptrI12rs2_pipelineE[_ZNK3rs26config11can_resolveESt10shared_ptrI12rs2_pipelineE]+0x3c): undefined reference to `rs2_config_can_resolve'
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::pipeline::pipeline(rs2::context)':
main.cpp:(.text._ZN3rs28pipelineC2ENS_7contextE[_ZN3rs28pipelineC5ENS_7contextE]+0x38): undefined reference to `rs2_create_pipeline'
/usr/bin/ld: main.cpp:(.text._ZN3rs28pipelineC2ENS_7contextE[_ZN3rs28pipelineC5ENS_7contextE]+0x44): undefined reference to `rs2_delete_pipeline'
/usr/bin/ld: main.cpp:(.text._ZN3rs28pipelineC2ENS_7contextE[_ZN3rs28pipelineC5ENS_7contextE]+0x48): undefined reference to `rs2_delete_pipeline'
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::pipeline::start(rs2::config const&)':
main.cpp:(.text._ZN3rs28pipeline5startERKNS_6configE[_ZN3rs28pipeline5startERKNS_6configE]+0x50): undefined reference to `rs2_pipeline_start_with_config'
/usr/bin/ld: main.cpp:(.text._ZN3rs28pipeline5startERKNS_6configE[_ZN3rs28pipeline5startERKNS_6configE]+0x5c): undefined reference to `rs2_delete_pipeline_profile'
/usr/bin/ld: main.cpp:(.text._ZN3rs28pipeline5startERKNS_6configE[_ZN3rs28pipeline5startERKNS_6configE]+0x60): undefined reference to `rs2_delete_pipeline_profile'
/usr/bin/ld: /tmp/ccpNhjFg.o: in function `rs2::pipeline::wait_for_frames(unsigned int) const':
main.cpp:(.text._ZNK3rs28pipeline15wait_for_framesEj[_ZNK3rs28pipeline15wait_for_framesEj]+0x38): undefined reference to `rs2_pipeline_wait_for_frames'
[... cut ...]
main.cpp:(.text._ZNK3rs28frameset10foreach_rsIZNKS0_16first_or_defaultE10rs2_stream10rs2_formatEUlNS_5frameEE_EEvT_[_ZNK3rs28frameset10foreach_rsIZNKS0_16first_or_defaultE10rs2_stream10rs2_formatEUlNS_5frameEE_EEvT_]+0x58): undefined reference to `rs2_extract_frame'
collect2: error: ld returned 1 exit status
```

I strongly assume that I made a mistake when linking the realsense2 library, since it seems to be missing the implementation, but i can't spot any error in my g++ command.

Thanks in advance :)

MartyG-RealSense, 2022-09-20:
Hi @mathklk  The SDK examples are built in a different way to how a typical user-created RealSense script is compiled.  The examples are installed as a set rather than individually.  They can be updated by modifying a particular program's script and then building the set of examples again.  It then builds only the programs that have been modified instead of the entire set of examples.  Further information about this can be found at https://github.com/IntelRealSense/librealsense/issues/6797

When taking code out of the examples and integrating it into your own project, this may therefore mean adding additional lines to CMakeLists.txt in order to avoid missing library linkages, as you found yourself with your own tests.  An example of this is the **Getting Started** self-contained tutorial project at the link below, which uses the same script as rs-hello-realsense but has a more complex CMakeLists.txt file.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

*******************

https://github.com/IntelRealSense/librealsense/issues/6182 is an example of trying to compile a standalone script with g++.  In that RealSense user's particular case they used the following instruction:

`$ g++ -std=c++11 record_camera.cpp -o record_camera -L/usr/local/include -L/usr/lib/x86_64-linux-gnu/ -lrealsense2 -lopencv_core -lopencv_imgproc`

I note that in their instruction, they are referencing **L/usr/local/include** and without " " quotations instead of -I"/usr/include/"

mathklk, 2022-09-20:
Hi @MartyG-RealSense , thanks for the reply, I found the mistake, I messed up the order of parameters given to g++. This is not actually a problem with rs2. **Source files have to come before libraries**, at least for my project structure.

This [stackoverflow thread](https://stackoverflow.com/a/14625444/11998115) contains a more detailed answer.

> The order of objects and libraries is relevant to the linker (called implicitly by the compiler when creating an executable). When the linker, in its left-to-right scan, finds a use of a name it doesn't know about it starts looking for a definition from that point on. If a definition passes by, it doesn't remember it for later use. -vonbrand

closed 2022-09-20 by mathklk

MartyG-RealSense, 2022-09-20:
It's great to hear that you achieved a solution, @mathklk - thanks very much for the update!