realsenseai/librealsense #10930: Connect Darknet node to realsense camera node

TheNemo05, 2022-09-22:
I am using YOLO V3 which uses darknet, trying to receive image from Intel realsense L515 camera. How do I connect /darknet node to /camera node as shown in the screenshot. I need to capture my realsense camera image into YOLO.
[image]

TheNemo05, 2022-09-28:
The above shared process is not working

TheNemo05, 2022-09-28:
@RealSenseSupport  please look into this

MartyG-RealSense, 2022-09-28:
I conducted extensive further research on this subject.  I was able to establsh that L515 can work with yolo3 and darknet_ros.  Unfortunately though there are not any better reference diagrams available than the one that I posted above.  I do apologize.

closed 2022-10-10 by TheNemo05