realsenseai/librealsense #10932: 2.43 Binary Installation

tugbakara, 2022-09-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version |   Ubuntu 18.04 LTS  | 
| Kernel Version (Linux Only)    |  4.9.253   | 
| Platform                                 | NVIDIA Jetson AGX Xavier  |
| SDK Version                            |  2.43                      | 
| Language                            |   | 
| Segment			|  Robot | 

### Issue Description
Hi @MartyG-RealSense , we have installed 2.43 from source but is there binary installation for this specific version? If it doesn't exist, which flags do I can use to install from source, thanks!

MartyG-RealSense, 2022-09-23:
Hi @tugbakara  A RealSense team member provides advice at https://github.com/IntelRealSense/librealsense/issues/5773#issuecomment-581477785 for installing a specific librealsense version from packages.

tugbakara, 2022-09-23:
Hi @MartyG-RealSense , we didn't try the binary installation but source installation seems working,thanks! :)

closed 2022-09-23 by tugbakara