realsenseai/librealsense #10933: [MSVC][std:c++latest] librealsense failed to build on MSVC

QuellaZhang, 2022-09-23:
Hi All,

The MSVC team recently added librealsense as part of RWC testing. And we found some errors when building librealsense with '/std:c++latest'. Could you please help look at this issue? Thanks in advance.

**Repro steps:**

1. open VS2019 x64 native tools
2. git clone https://github.com/IntelRealSense/librealsense.git F:\gitP\IntelRealSense\librealsense
4. cd F:\gitP\IntelRealSense\librealsense
5. mkdir F:\gitP\IntelRealSense\librealsense\build_amd64
6. cd F:\gitP\IntelRealSense\librealsense\build_amd64
7. cmake -G "Visual Studio 16 2019" -A x64 -DCMAKE_SYSTEM_VERSION=10.0.18362.0 -DCMAKE_BUILD_TYPE=Release -DBUILD_LEGACY_LIVE_TEST=true -DBUILD_EXAMPLES=true -DBUILD_WITH_TM2=true -DCHECK_FOR_UPDATES=true ..
8. msbuild /m /p:Platform=x64 /p:Configuration=Release librealsense2.sln /t:Rebuild

**Error info:**
F:\gitP\IntelRealSense\librealsense\src\media\record\record_sensor.cpp(216,97): error C2385: ambiguous access of 'recordable' [F:\gitP\IntelRealSense\librealsense\build_amd64\realsense2.vcxproj]
F:\gitP\IntelRealSense\librealsense\src\mf\mf-uvc.cpp(1133,24): error C2440: 'initializing': cannot convert from '_InIt' to '_InIt &' [F:\gitP\IntelRealSense\librealsense\build_amd64\realsense2.vcxproj]

MartyG-RealSense, 2022-09-23:
Hi @QuellaZhang  My first thught is that the problem may be with the **c++latest** instruction.  The librealsense SDK is targeted at **c++11**, as referenced by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/6283#issuecomment-645490414

You may also be able to build librealsense from source code with a simpler CMake instruction:

**cmake ../ -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true**

QuellaZhang, 2022-09-27:
> Hi @QuellaZhang My first thought is that the problem may be with the **c++latest** instruction. The librealsense SDK is targeted at **c++11**, as referenced by a RealSense team member at [#6283 (comment)](https://github.com/IntelRealSense/librealsense/issues/6283#issuecomment-645490414)

Yes, all these errors are caused by the **c++latest** instruction.

> You may also be able to build librealsense from source code with a simpler CMake instruction:
> **cmake ../ -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true**

@MartyG-RealSense, I tried simpler CMake instructions and the error still exist. We made a simple patch, can can help review?
[librealsense_error_c2385.patch.txt](https://github.com/IntelRealSense/librealsense/files/9652118/librealsense_error_c2385.patch.txt)

MartyG-RealSense, 2022-09-27:
Thanks very much for the patch.  I will discuss the issue with my Intel RealSense colleagues.  Thanks again!

MartyG-RealSense, 2022-10-14:
Case closed due to no further comments received.

closed 2022-10-14 by MartyG-RealSense

QuellaZhang, 2022-12-09:
@MartyG-RealSense, ohhhhh, I am very sorry and I know what happened here. I missed **step7**, that's why you guys can't reproduce the problem. Could you please try again? Thank you.

Or you can use the simpler command **set CL=/std:c++latest** which will expose more other errors.

Sorry again😢.

MartyG-RealSense, 2022-12-09:
It's no trouble at all, @QuellaZhang - I will pass your new information to my colleagues.  Thanks!

MartyG-RealSense, 2022-12-10:
With your new information the errors were able to be reproduced.  I will update you when I have further information to provide.  Thanks again!