realsenseai/librealsense #10935: Convert ROI points from color coordinates to depth (for exposure)

samsont, 2022-09-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Operating System & Version |   Ubuntu 20  | 
| Platform                                 | PC |
| SDK Version                            |  2           | 
| Language                            |  python                          | 

### Issue Description
What is best practices for converting a ROI coordinates from color coordinates to depth coordinates? 

In particular, I'd like to set a ROI for exposure control in both color and depth coordinates. I run an object detector on the color image, get a bounding box, and I want to set the ROI for both. 

I currently using the align command to map the depth image (1280x720) onto the color image (1920x1080) so that I can work in the color image space. That's where I get my bounding box.  I'm presume I need to map that bounding box back into the un-aligned depth image, in the 1280x720 image. I'm not sure how to do that.

I see in https://github.com/IntelRealSense/librealsense/issues/6803, that @MartyG-RealSense and @dorodnic recommend using rs2:align. I believe the align command aligns images.   How does one align points, the 4 corners of the bounding box? Thanks!

MartyG-RealSense, 2022-09-23:
Hi @samsont  Unless you are experiencing image problems related to the sensors being saturated with sunlight when using a camera outdoors in bright light, you may not need to set a ROI on the RGB.  If your project requires one then https://github.com/IntelRealSense/librealsense/issues/3427 provides Python code for defining an RGB ROI.

A ROI can be separately defined for both depth and RGB by including code in the script to inform it to access the depth sensor or the RGB sensor depending on which type of ROI you want to set.  So you would not need to convert a ROI from color to depth or vice versa.

If you are using **rs2::align** in your script to perform the depth to color alignment then the SDK's **align processing block** will make automatic adjustments for differences in resolution between depth and color.

samsont, 2022-09-23:
@MartyG-RealSense . Thanks for your quick response, and for your pointers on setting an ROI, and your comments on sunlight. 

I'm a bit confused on why I would _not_ need to convert an ROI from color to depth.

If I follow the code an #3427 to set a color ROI, and then I want to set a corresponding ROI on the depth sensor,  I would imagine that I don't send in the exact same coordinates to the code in #2681.  After all, that sensor has a different resolution and different coordinates. And, while I might align a particular frame, the sensor behaves in its own coordinates.  Or do I not understand correctly?

To be precise: let's say I knew I wanted an ROI at (1820,980) to (1920,1080) in the 1920x1080 color sensor using  #3427. And, let's say I want to set a corresponding ROI in the depth sensor using the code you pointed at in #2681. I would think that the coordinates need to be converted in some way to set the ROI on the depth sensor.

MartyG-RealSense, 2022-09-24:
The RGB ROI is a separate option from the depth ROI that does not affect depth image quality, only the color image quality.  The depth and RGB ROIs are independent of each other.

Depth to color alignment will map a depth coordinate onto the corresponding RGB coordinate on the aligned image.  So you could visualize it as two different pixel types that are independently affected by their particular ROIs (which do not interact with one another) but overlaid on one another.  Remember that the SDK's 'align processing block' automatically adjusts for differences in resolution when aligning one stream type to another.

I believe that each individual image type (depth and color) has its pixels adjusted by that image type's ROI, and then the pixels of the two images are aligned.  So it is not aligning the ROIs, but the pixels in the areas of the image that have been affected by the ROIs prior to alignment.

***********

If you wanted to perform alignment the other way around then you could utilize the much more rarely used **color to depth** alignment technique so that a color coordinate is mapped onto a depth coordinate.  To perform color to depth alignment you can simply change the **align_to** instruction to align to depth instead of color.  For example:

```
align_to = rs.stream.depth
align = rs.align(align_to)
```

MartyG-RealSense, 2022-09-29:
Hi @samsont  Do you require further assistance with this case, please?  Thanks!

samsont, 2022-09-29:
@MartyG-RealSense : As you point out, I can do the color to depth alignment.  Since I have coordinates in the un-aligned color image, is there a way to figure out to where those coordinates moved?  Thanks.

MartyG-RealSense, 2022-09-29:
After alignment, you could obtain the Z-depth coordinate by using **rs2_deproject_pixel_to_point** like the script at https://github.com/IntelRealSense/librealsense/issues/2458

closed 2022-10-01 by samsont