realsenseai/librealsense #10936: Using pipeline callback function causes 100% cpu load

mathklk, 2022-09-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |  Linux arm64 (Raspberry Pi 4) | 
| Kernel Version (Linux Only)    |  5.15.61-v8+                                         | 
| Platform                                 | Raspberry Pi 4 Model B Rev 1.4 |
| SDK Version                            |  2.44.0                          | 
| Language                            |  C++                        | 
| Segment			|  others                          | 

### Issue Description

I'm using the following code to read frames from my camera in one thread and process them in another thread. Im using a callback to write the incoming frames to a frame queue, as described in the docs under [frames and threads](https://dev.intelrealsense.com/docs/frame-management#frames-and-threads).

```cpp
/* this happends in the main() function*/

rs2::frame_queue q;
rs2::pipeline pipeline;
rs2::config config;
config.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 15);

pipeline.start(config, [](rs2::frame f){
	q.enqueue(f);
});

// afterwards I create a processing thread using std::thread, that reads from the queue
```

The program works, but I noticed that one of my four CPU cores is at 100% load. When I disable my working thread the problem persists. When I disable the callback function, CPU load is back to normal. 

So my question is, am I doing something wrong or is this the normal behaviour of using the callback function? Are there better ways to read frames that do not cause 100% load?

MartyG-RealSense, 2022-09-23:
Hi @mathklk  Could you first try inserting the word **callback** into the brackets of your pipeline.start instruction please so that the SDK knows that it is a callback script?

```
pipeline.start(config,callback,[](rs2::frame f){
q.enqueue(f);
```

An example of this is demonstrated in the pipeline start line of the RealSense SDK's C++ **rs-callback** program.

https://github.com/IntelRealSense/librealsense/blob/master/examples/callback/rs-callback.cpp#L46

mathklk, 2022-09-27:
This turned out to not be an issue with the pipeline callback function, but something else entirely.

closed 2022-09-27 by mathklk