realsenseai/librealsense #10937: Undistort Matlab

jdanielhoyos, 2022-09-23:
Hello everyone. I'm tring to undistort the image in matlab but I only found how todo it in python. https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py
Any material?
Thanks

MartyG-RealSense, 2022-09-24:
Hi @jdanielhoyos  MATLAB has a function called **undistortImage**.  

https://uk.mathworks.com/help/vision/ug/remove-distortion-from-an-image-using-the-cameraparameters-object.html

A RealSense user with a 400 Series camera at https://github.com/IntelRealSense/librealsense/issues/10363#issuecomment-1086717404 to undistort with the instruction below.

`[J,~] = undistortImage(imread(right_images(i)), params.params.CameraParameters2);`

jdanielhoyos, 2022-09-24:
Hello @MartyG-RealSense . I did my calibration using the calibrator app, but as it is mentioned in the post you cite, it is not ideal, so when I do the disparity map, it is just a mess. Is there an official calibration parameters for params.params.CameraParameters2? Thanks

MartyG-RealSense, 2022-09-24:
A RealSense user at https://github.com/IntelRealSense/librealsense/issues/2770 who used the MATLAB calibrator app also had problems.

A RealSense camera's complete intrinsics and extrinsics can be obtained using the RealSense SDK's **rs-enumerate-devices** tool in calibration information mode by launching the tool with the instruction below.

**rs-enumerate-devices -c**

closed 2022-10-07 by MartyG-RealSense