realsenseai/librealsense #10940: How to convert RS2_FORMAT_BGR8 infrared channel to RS2_FORMAT_Y8 infrared channel?

NingCY, 2022-09-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Win10| 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC|
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I usually use cfg.enable_stream(RS2_STREAM_INFRARED, WIDTH, HEIGHT, **RS2_FORMAT_Y8**, FRAME_RATE) to enable ir stream, I find 
cfg.enable_stream(RS2_STREAM_INFRARED, WIDTH, HEIGHT, **RS2_FORMAT_BGR8**, FRAME_RATE) can also return one image normally and have 3 channels, how can I convert the **RS2_FORMAT_BGR8** one to **RS2_FORMAT_Y8** one?

NingCY, 2022-09-26:
I know ir image should be a 8-bit gray image, but if I set **RS2_FORMAT_BGR8** to ir stream, what will happen? How can I know exactly what's going to happen?  Can anyone give some advice please? Thanks!

MartyG-RealSense, 2022-09-26:
Hi @NingCY  You can obtain an RGB color image from the left infrared sensor instead of a Y8 infrared image on some RealSense camera models (D405, D415 and D455) but you cannot convert a live RGB8 / BGR8 color stream to a Y8 infrared stream unfortunately.

RGB images from the left infrared sensor will have the same viewpoint as a Y8 image from the left sensor as they are both generated from the same sensor.

*****

If you are instead aiming to export an 8-bit gray image to an image file then you can use OpenCV code such as the C++ scripting at the link below to export the Y8 infrared image to PNG.

https://github.com/IntelRealSense/librealsense/issues/1480

If you disable the camera's IR emitter in your code to remove the dot pattern that the projector casts onto objects in the scene then an exported PNG of the Y8 stream will closely resemble a monochrome RGB image when the dot pattern is absent.

NingCY, 2022-09-26:
Thanks for your detailed information. I‘m using D415 and I may need to ask several questions to confirm I understand right:
1. The RGB color image from the **left infrared sensor** don't have relationship with Y8 infrared image also from the **left infrared sensor**. They have the same viewpoint but can not convert to each other online(live stream) or offline(saved image), right?
2. About IR emitter you mentioned, I try and get ir image without the dot pattern, but the accuracy of the depth image drops significantly, so the IR emitter can enhance depth image, and I can't get accurate depth image and no dot pattern ir image at the same time, is it true? 
3. About **auto-exposure**, I'm in an office with normal indoor light condition, I find that if I open **auto-exposure**, the ir images will become darker, if I close it, it will become brighter, how does it works?
Looking for your relpy and thanks in advance.

NingCY, 2022-09-26:
I'd like to get brighter ir images because it have more information, so should I close the **auto-exposure**?

MartyG-RealSense, 2022-09-26:
1.  You are correct.  The RGB color from left infrared sensor and Y8 infrared are separate formats that are not convertible from one to the other but have the same viewpoint.

2.  RealSense 400 Series cameras can also use the ambient light in a scene to analyze surfaces for depth detail instead of using the dots as a 'texture source' to analyze.  So although depth detail can reduce if the dot pattern is disabled, increasing the lighting level of the scene if it is possible to do so may compensate for the pattern's absence.

Alternatively, there is a camera configuration Visual Preset for the D415 model only called **Left Imager Color w/o IR Pattern** that can remove the dot pattern without disabling the emitter.

https://dev.intelrealsense.com/docs/d400-series-visual-presets#preset-table

3.  When Autoexposure is turned on, it will average the intensity of all the pixels inside of a predefined Region of Interest (ROI) and will try to maintain this value at a predefined Setpoint.  Both the ROI and the Setpoint can be set in software.  In the RealSense Viewer the setpoint can be found under **Stereo Module > Advanced Controls > AE Control**.  Increasing the Setpoint value above its default of '1000' will brighten the infrared image without having to disable auto-exposure.

For example, changing the setpoint from 1000 to 2000 and 3000 had the following effect on the infrared image in a test that I performed.

**1000**

[image]

**2000**

[image]

**3000**

[image]

NingCY, 2022-09-27:
Thanks for your patient reply. 
May I know the default value of predefined Region of Interest (ROI) without manual setting? Because I have multiple scenes, ROI position sometimes unable to know, so I want to know if exists a generic ROI setting?

MartyG-RealSense, 2022-09-27:
The 'Default' camera configuration Visual Preset files suggest the following defaults for the Setpoint value based on the camera model being used.

D415: 400
D435: 1536
D455: 1000

https://dev.intelrealsense.com/docs/d400-series-visual-presets#preset-table

NingCY, 2022-09-27:
> The 'Default' camera configuration Visual Preset files describe the following defaults for the Setpoint value based on the camera model being used.
> 
> D415: 400 D435: 1536 D455: 1000
> 
> https://dev.intelrealsense.com/docs/d400-series-visual-presets#preset-table

Yeah, I know, I can see from Intel RealSense Viewer, the Setpoint is 400 using D415 see as below:
[image]

> 3. When Auto-Exposure is turned on, it will average the intensity of all the pixels inside of a predefined Region of Interest (ROI) and will try to maintain this value at a predefined Setpoint.

What I exactly want to ask is the default setting of **ROI region**,  the range of all pixels used to calculate the average value. Is default ROI the whole images(I guess) if I have not set manually?

MartyG-RealSense, 2022-09-27:
By default the ROI will be centred in the middle of the image.  On the 400 Series cameras the data sheet document defines the ROI area as 80% of the field of view rather than the whole image.

closed 2022-10-09 by MartyG-RealSense