realsenseai/librealsense #10943: RS D455 Cannot Be Detected

cjh3020889729, 2022-09-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455  | 
| Firmware Version                   | latest | 
| Operating System & Version |   Linux(JetPack4.x)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 |  NVIDIA Jetson  |
| SDK Version                            |  2.50.0 }                          | 
| Language                            |  Cpp           | 
| Segment			|  others                      | 

### Issue Description
Firstly, I have successfully compiled 2.50.0 SDK on the board.
I inserted the USB of D455 into the USB interface of Jetson Xavier, but Jetson could not detect the camera's access.When I use lsusb to view, no change before and after insertion.
Can you tell me about this situation? Thanks.

Note: When I insert it on the notebook, I can use Realsense normally. (on windows)

cjh3020889729, 2022-09-27:
Add:
When I first connect to an ordinary USB camera on the board, Realsense can be successfully detected. But use the program of rs-viewer, it show the connection is 2.0 usb.

MartyG-RealSense, 2022-09-27:
Hi @cjh3020889729  Which method did you use to install the librealsense SDK on your Jetson, please?

cjh3020889729, 2022-09-27:
First of all thank you for your reply.
Sorry，I am not very clear，which the SDK compilation is done by my colleague. But it can be used well.
When I use the "Add" method mentioned above, the realsense camera can get deep images normally.

Do you mean, maybe some dependencies are missing before compiling SDK?

MartyG-RealSense, 2022-09-27:
If you can add the camera using the **Add Source** option then this is a valid way of listing a RealSense camera in the Viewer if it is not appearing automatically when plugged in, as Add Source is the method by which more than one RealSense camera can be added to the camera list in the options side-panel.

When using Add Source, do you have the 'Intel RealSense D455' option available and select that?

[image]

cjh3020889729, 2022-09-27:
Ok, I will to try and look at it.
After I test it, I will reply to you.
Thanks.

MartyG-RealSense, 2022-09-28:
You are very welcome.  Thanks very much for your update and sharing the findings of your tests!

closed 2022-10-09 by MartyG-RealSense