realsenseai/librealsense #10944: Intel Realsense D4XX recovery for D450

reals2018, 2022-09-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D450 } | 
| Firmware Version                   | (5.12.15.50) | 
| Operating System & Version |   {Win (10)  | 
| Platform                                 | PC |
| SDK Version                            |  { 2.4.9 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I am trying to connect D450 with D4 board V1, in the D400 series  datasheet

(1) The V1 ASIC (without IMU) is validated and supports all D400 depth modules

but it shows Intel Realsense D4XX recovery in the Realsense Viewer 

physicial port: \\?\USB#VID_8086&PID_0ADB#111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
firmware update id:118223024220

it works ok for V3 board, so it might not be the issue of  interposer.

any idea how to solve this?

MartyG-RealSense, 2022-09-28:
Hi @reals2018  The D450 depth module must be used with the Vision Processor D4 V3 board.  Likewise, the D401 module should only be paired with the V4 board.  I will refer the (1) note to my Intel RealSense colleagues.  Thanks!

MartyG-RealSense, 2022-10-05:
Hi @reals2018  Do you require further assistance with this case, please?  Thanks!

reals2018, 2022-10-05:
Not now. I will close this first.Thanks for the help

closed 2022-10-05 by reals2018