realsenseai/librealsense #10949: Question - Facial tracking with Intel D415 ( C# SDK Version 2.38)

rajrajeswaran, 2022-09-29:
Camera Model: Intel D415
Firmware: 05.12.07.100
Operating System: Windows 10
Platform: PC
SDK version: C# RealSense SDK 2.38
Language: C#
Segment: Windows Forms application.

I am currently using Intel RealSense C# SDK version 2.38 with Intel D415 sensor.  Using RealSense viewer, I am able to track faces.  Does the C# SDK also support facial detection and tracking?  If so, is there a sample application that i can look at?  Essentially, I am looking to blur faces detected by the sensor in a C# windows forms application.

MartyG-RealSense, 2022-09-29:
Hi @rajrajeswaran  I believe that you are referring to the **Face Detection: OpenVINO** RGB post-processing filter in the RealSense Viewer.  Is that correct, please?

[image]

This face detection system integrates the **OpenVINO** vision computing platform into the Viewer to provide facial analysis.  This feature cannot be accessed directly with scripting outside of the Viewer, but the RealSense SDK is able to make use of OpenVINO via its compatibility wrapper at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openvino

The wrapper's code is based on C++ rather than C#.  There is not an available example of accessing the RealSense OpenVINO wrapper from C#, unfortunately.

In regard to alternative approaches to OpenVINO, if you google for the term **opencv blur face c#** then a range of approaches for using OpenCV (which is fully compatible with the RealSense SDK) to isolate and blur faces are suggested in the search results.

MartyG-RealSense, 2022-09-29:
Thanks very much @rajrajeswaran - as you suggested, I will close the case.  Thanks again!

closed 2022-09-29 by MartyG-RealSense