realsenseai/librealsense #10950: Focal Length Calibration - Failed to extract target information

drCJH, 2022-09-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu 18.04.6  | 
| Kernel Version (Linux Only)    | 4.15.0-189                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.51.1.4348                          | 
| Language                            |  python                          | 
| Segment			|  Robot                           | 

I am trying to calibrate my D435 using the depth_auto_calibration_example.py script from the Python examples folder, and yet every time the focal length calibration fails with the message 'RuntimeError: Failed to extract target information from the captured frames!'

 I have printed the recommended calibration target and positioned it on a wall in the center of the camera's view, and tried positioning the camera at different distances from 0.5 to 1.5m from the wall.

Here is an example image from the right IR sensor positioned 1m from the wall:
[image]

Here is the complete terminal output from running the script:

> Starting UCAL...
Starting On-Chip calibration...
        Speed:  wall
        Scan:   intrinsic
On-Chip calibration finished
        Health: -0.054392170161008835
Starting Focal-Length calibration...
        Target Size:    (175, 100)
        Side Adjustment:        right_only
Traceback (most recent call last):
  File "calibration.py", line 285, in <module>
    main()
  File "calibration.py", line 88, in main
    run_focal_length_calibration((args.target_width, args.target_height), args.focal_adjustment)
  File "calibration.py", line 176, in run_focal_length_calibration
    fl_adjust_map[adjust_side],progress_callback)
RuntimeError: Failed to extract target information
from the captured frames!

MartyG-RealSense, 2022-09-30:
Hi @drCJH  Are you able to instead test focal length calibration in the RealSense Viewer tool to see whether calibration still fails with the target image, please?  The focal length calibration can be tested in the Viewer by left-clicking on the **More** option near the top of the Viewer's options side-panel and selecting **Focal Length Calibration** from its drop-down menu.

[image]

drCJH, 2022-09-30:
The viewer returns the same 'Failed to extract target information' error message.

MartyG-RealSense, 2022-09-30:
Thank you very much for testing in the Viewer.  Does it make a difference if the target image is moved higher up the wall and taped at both top and bottom so that it is positioned level, flat against the wall and does not have the dark base of the wall in the camera's field of view?

MartyG-RealSense, 2022-09-30:
I conducted tests with a D435 and the calibration image.  I found that the **Failed to extract target information** error was most likely to occur if:

-  The lighting level around the calibration target image was low.

-  The boundary box was not aligned with the target image when the calibration was activated.  Activating the calibration with the boundary box not aligned around the edges of the target image on the wall and then moving the camera to align the box with the target would result in the error.  

The easiest way to achieve alignment of the boundary box with the image from the beginning if you do not have a surface to rest the camera on is to align the boundary box to the image after calibration starts (which will result in a fail) and then hold the camera as still as possible in the hand and click 'retry' with the other hand so that the boundary box is aligned with the target from the beginning on the next attempt.

drCJH, 2022-10-03:
Thanks, I managed to get it to work. I think the problem was that I was too close to the target and so it wasn't fully inside the ROI of both sensors. By moving back to ~80cm I was able to get the target in both ROIs and complete the calibration.

closed 2022-10-03 by drCJH

MartyG-RealSense, 2022-10-03:
It's great news that you succeeded, @drCJH - thanks very much for the update!