realsenseai/librealsense #10954: Help reviving a bricked Intel RealSense L515

ferrolho, 2022-10-02:
I am looking for help regarding a broken _Intel RealSense L515_.

| Required Info               |                                            |
| --------------------------- | ------------------------------------------ |
| Camera Model                | L515                                       |
| Firmware Version            | `1.5.8.1`, `1.5.5.0`, `1.5.4.1`, `1.5.3.0` |
| Operating System & Version  | Windows 10, Linux (Ubuntu 20.04)           |
| Kernel Version (Linux Only) | `5.15.0`                                   |
| Platform                    | PC                                         |
| SDK Version                 | `2.51.1`, `2.49.0`, `2.48.0`               |
| Language                    | `realsense-viewer`                         |

### Issue Description

It's a bit of a long story, so let's start!

I was using Ubuntu 20.04 and the RealSense Viewer (_librealsense_ `v2.51.1`) when the issues started to show up. Suddenly, I was no longer able to use any of the `realsense` tools to talk to the camera. This is the error I would get:
```
RealSense error calling rs2_create_device(info_list:0x562a6210e740, index:0):
    hwmon command 0x2b( 4 0 0 0 ) failed (response -29= Value out of range)
 01/10 23:26:37,500 WARNING [139734253248832] (rs.cpp:310) null pointer passed for argument "device"
```
> This same error would show up in `rs-fw-update -l`, in `realsense-viewer`, and in other `rs-*` command line tools.

At this point, I thought: is it my Ubuntu installation acting up? Perhaps even the cable? Or is it really the L515? To understand this, I plugged another L515 camera to the same computer and using the same cable; this time, no errors showed up. So the issue must have really been with that first L515 camera.

I decided to look into the issue a bit further, and so I switched over to my Windows 10 installation, where I have multiple _Intel.RealSense.Viewer.exe_ and _Intel.RealSense.FW.Update.exe_ using different versions of _librealsense_ (which I downloaded from the GitHub Releases assets.

I started by plugging the camera to the computer running Windows 10, and then confirmed the camera was being detected by Windows. I opened the Device Manager and, under Cameras, I could indeed see the L515 listed as:
- Intel(R) RealSense(TM) 515 Depth
- Intel(R) RealSense(TM) 515 RGB

Knowing that Windows was recognising the camera, I started launching different versions of the RealSense Viewer, trying one at a time, starting from the most recent version (`v2.51.1`) and making my way to the older versions. Suddenly, when I reached (`v2.48.0`), the camera started being detected again. I tried it multiple times just to double-check, and I can confirm:
> **the camera was being detected by the _librealsense_ tools (including the viewer) up to version `v2.48.0`, and then stopped working from `v2.49.0`–`v2.51.1`.**

At this point, I launched the RealSense Viewer to check the firmware running on the camera: it was `v1.5.8.1`. Still within the RealSense Viewer, I enabled the "Motion Module" to see the accel and gyro streams: they were working, and showing plausible values. I tried enabling also the "RGB Camera", and sure enough the color stream showed up and working well. Finally, I tried to enable the "L500 Depth Sensor", but this failed to launch and none of the depth, infrared, or confidence streams managed to show up. I tried to use the RealSense Viewer option for doing a "Hardware Reset", but this did not solve the problem.

At this point, I thought maybe the problem would go away if I were to reinstall the camera firmware, so I used `rs-fw-update` to reinstall the firmware version `1.5.8.1`. The terminal output showed 100% success, and I decided to give the camera another go, this time using again the latest version of the RealSense Viewer available to date. However, no joy there; all versions `v2.49.0`–`v2.51.1` were still not detecting the camera. So I launched RealSense Viewer `v2.48.0` and the camera was there, with the rgb and motion streams working, but still no luck with the depth streams.

Then, I decided to try installing older versions of the camera firmware. My approach was similar to the one I had used for the viewer: I started by installing the most recent firmware, and then going back one version at a time. I downgraded to `v1.5.5.0` but nothing changed. Then I downgraded to `1.5.4.1`, and still nothing changed. Finally, I installed firmware `1.5.3.0`, and something did change ─ unfortunately, for the worst.

After downgrading the camera firmware to version `1.5.3.0`, I was no longer able to detect the camera in RealSense Viewer (not even in the older version `2.48.0` that had worked before). I opened up the Windows Device Manager again, and to my surprise, the L515 was no longer being detected. I went back to Ubuntu, where I used `lsusb` to confirm it: the camera was no longer being detected _at all_.

So, that's the story about this problematic L515. In summary, it was being detected by different OS (windows and ubuntu), but not by the latest realsense tools. Then, after trying an older version of the realsense tools, the camera was detected (where the rgb and motion modules were working, but the depth module was not). Finally, after downgrading the camera firmware, the camera became completely unreachable: not even windows or ubuntu could detect its USB interface.

Is there anything that I can do to try to bring the camera back to life? Thank you in advance.

MartyG-RealSense, 2022-10-02:
Hi @ferrolho   Please try unplugging the micro-sized end of the USB cable from the base of the L515 camera, turning the connector around the opposite way and re-inserting it into the base of the camera (USB-C cables are two-way insertion at the micro-sized end).

Can the depth stream be detected after reversing the direction of the connector?

ferrolho, 2022-10-02:
Hi, Marty. Thank you for your prompt reply. Unfortunately, the camera seems to be completely bricked now, and I can't get the computer to detect it no matter which way I insert the cable. But I should've mentioned in my original post that I did try to insert the camera using more than one USB 3.1 cable, and I did try to plug the end that plugs into the camera both ways ─ it did not make a difference.

MartyG-RealSense, 2022-10-02:
Is the camera being plugged into a USB 3 port on the **computer**?  Originally at launch the L515 was USB3 only.  USB 2 support was added in a later firmware update.  If the firmware was downgraded to one that only had USB 3 support, then the camera would need to be inserted in a USB port that detected the camera as being on a USB 3 connection in order to recognize the camera and be able to upgrade the firmware to one that supports a USB 2 connection.

ferrolho, 2022-10-02:
> Is the camera being plugged into a USB 3 port on the computer?

Yes, it is.

MartyG-RealSense, 2022-10-02:
A camera connection may sometimes be mis-detected as USB 2 even when plugged into a USB 3 port.  Do you recall please whether the number beside the camera name in the Viewer was 2.1 or 3.2 when the camera was able to be detected in the Viewer, please?  If it was 2.1 then this would indicate that the USB connection was being mis-detected as USB 2.  This could occur if the USB micro connector was inserted a certain way around of the two orientations available.

ferrolho, 2022-10-02:
When the camera was able to connect to the RealSense Viewer, the USB connection was showing as version 3.2, not 2.1.

MartyG-RealSense, 2022-10-02:
I note that you were able to use another L515 on the same computer with no problems, suggesting that the detection issue is specific to the specific camera affected rather than a problem with the computer.

In regard to enabling RGB, depth and RGB, there is a known issue where enabling these three streams in the Viewer at the same time can cause data on one of the streams to cease being received.  The trigger for that problem is the enabling of the Motion Module.  A hardware reset does not correct this and a power-off and power-on of the computer is required to clear the problem.  It is unlikely that this issue would prevent the camera from being detected though.

Is the camera still undetectable if you download for Windows the 2.41.0 version of **Intel.RealSense.Viewer.exe**, as this SDK version matches to L515 firmware version 1.5.3.0.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.41.0

Also, please check the USB categories of the Windows Device Manager to see whether the RGB and Depth drivers are listed there instead of in the usual 'Cameras' category.

ferrolho, 2022-10-02:
I downloaded the version of the RealSense Viewer that you suggested ─ no luck there.

Unfortunately, I don't think the camera will ever be detected by the RealSense Viewer or any RealSense command line tool if the operating system is not able to detect the camera. I tried to connect it to Windows and to Ubuntu (and also tried different computers), but the camera is not detected.

When I use the "working L515", there's also the familiar sound notification that Windows plays when a device is connected/disconnected, whereas no sound is played when I connect/disconnect the "broken L515".

Here is what the Device Manager shows on Windows:
| Working L515                                                                                                                          | Broken L515                                                                                                                           |
| ------------------------------------------------------------------------------------------------------------------------------------- | ------------------------------------------------------------------------------------------------------------------------------------- |
| [image] | [image] |

MartyG-RealSense, 2022-10-02:
You have tested the affected camera thoroughly enough to demonstrate that the camera is likely unrecoverable.  If you purchased the camera within the past 12 months then please contact the retailer that you purchased it from to enquire about a return.

ferrolho, 2022-10-02:
Okay. Thank you for your help, Marty.

closed 2022-10-02 by ferrolho