realsenseai/librealsense #10955: Intel RealSense L515 producing inaccurate depth information

ferrolho, 2022-10-02:
Hello! I am looking for help regarding a (possibly) faulty depth module of an Intel RealSense L515.

In the tables below, I compare data from two different Intel RealSense L515, `f0270351` and `f0350132`.

> Disclaimer: I did *not* run the cameras at the same time, so there is no way one camera's IR would interfere with the other.

| Required Info               |       f0270351       |       f0350132       |
| --------------------------- | :------------------: | :------------------: |
| Camera Model                |         L515         |         L515         |
| Firmware Version            |      `1.5.8.1`       |      `1.5.8.1`       |
| Operating System & Version  | Linux (Ubuntu 20.04) | Linux (Ubuntu 20.04) |
| Kernel Version (Linux Only) |       `5.15.0`       |       `5.15.0`       |
| Platform                    |          PC          |          PC          |
| SDK Version                 |       `2.51.1`       |       `2.51.1`       |
| Language                    |  `realsense-viewer`  |  `realsense-viewer`  |

This table shows the configuration for each camera in RealSense Viewer.

As shown in the screenshots, the configuration parameters are set to the exact same values.

|                                                         f0270351                                                         |                                                         f0350132                                                         |
| :----------------------------------------------------------------------------------------------------------------------: | :----------------------------------------------------------------------------------------------------------------------: |
| [image] | [image] |

To compare the output of both cameras, I set them up next to each other, and then I placed a box 60 cm away from the cameras. The picture below shows the experiment set up.

> Please note that the ruler and the target box in the picture below are aligned for the first camera, but then were moved and realigned to collect the data for the other camera.

[image]

Looking just at the depth streams (shown below) does not seem to show much difference between one camera and the other:

|                                                        f0270351                                                         |                                                        f0350132                                                         |
| :---------------------------------------------------------------------------------------------------------------------: | :---------------------------------------------------------------------------------------------------------------------: |
| [image] | [image] |

However, looking at the IR streams, we start to see some differences. Here, the stream from the camera on the left seems much stronger than the one on the right, which looks fainter (even though both cameras are set to the same laser strength and receiver gain).

|                                                        f0270351                                                         |                                                        f0350132                                                         |
| :---------------------------------------------------------------------------------------------------------------------: | :---------------------------------------------------------------------------------------------------------------------: |
|  [image]   |  [image]   |

Next, I placed my cursor in the same position (tip of letter "t" in the Intel logo).

As you can see in the pictures below, one of the cameras computes the depth as 0.608 m whereas the other one computes 0.570 m. In the real world, the box was set at approximately 60 cm from each camera. That means the error in the first one is ~0.8 cm, and 3.0 cm in the second one.

The error of the first camera (under 1 cm) seems reasonable (correct me if I'm wrong), but the error of the second camera seems unreasonable. Anyway, the fact that these cameras output a value which is almost 4 cm apart seems unreasonably inconsistent...

|                                                        f0270351                                                         |                                                        f0350132                                                         |
| :---------------------------------------------------------------------------------------------------------------------: | :---------------------------------------------------------------------------------------------------------------------: |
|  [image]  |  [image]  |

For the next experiment, I wanted to highlight the other inconsistent aspect between one camera and the other: the quality of the point cloud on a flat surface. To do this, I pointed both cameras at my ceiling.

Below you can find the screenshots of the raw point cloud from both cameras. Notice how the one on the right is much more noisy than the one on the left. (And trust me: my ceiling is quite flat.)

|                                                             f0270351                                                             |                                                             f0350132                                                             |
| :------------------------------------------------------------------------------------------------------------------------------: | :------------------------------------------------------------------------------------------------------------------------------: |
|    [image]     |    [image]     |

Finally, these last two screenshots show the 3D view of the depth + rgb streams. Again, I am hovering my cursor approximately at the same point, but the cameras output two very different values: 1.603 m vs 1.562 m.

|                                                             f0270351                                                             |                                                             f0350132                                                             |
| :------------------------------------------------------------------------------------------------------------------------------: | :------------------------------------------------------------------------------------------------------------------------------: |
| [image] | [image] |

Is it normal that L515 cameras can be this inconsistent from one unit to the other? And is there anything I can do to fix the issue with the depth stream quality?

Thank you in advance.

MartyG-RealSense, 2022-10-03:
Hi @ferrolho  The most noticable difference in the settings that I observed is that camera **f0270351** seems to have significantly higher internal temperatures than the other camera.  High temperatures can cause depth readings to change.  https://github.com/IntelRealSense/librealsense/issues/8434 is a case that provides an example of this on 400 Series cameras.

If both cameras are attached to the same computer then I would recommend first trying to swap the USB cables of the two cameras in order to confirm whether the f0350132 camera has the same deviation in depth measurement when using the cable of the f0270351 camera with the elevated temperature.  

If the measurement deviations occur on the f0350132 camera when using f0270351 's cable then it might indicate a bad / damaged USB cable that is causing the camera to overheat.

ferrolho, 2022-10-03:
Hi, Marty. Thank you for your reply.

---

> camera **f0270351** seems to have significantly higher internal temperatures than the other camera

This is because, at the time of the screenshot, **f0270351** had been plugged in for a while, whereas the config screenshot for **f0350132** was taken right after plugging it in.

Anyway, the camera with the bad depth is **f0350132**, and the problem does not change over time with temperature. In other words, the depth stream of **f0350132** is as bad as shown in my pictures above, even straight after plugging it in (with low temperatures). And the depth stream of **f0270351** (the "good" camera) is good even at the "high" temperatures shown in the config side panel.

---

> Changes in temperature can cause depth readings to change. https://github.com/IntelRealSense/librealsense/issues/8434 is a case that provides an example of this on 400 Series cameras.

I see... that is interesting! But the L515 uses a different system from the 400 Series, right? The L515 is the only RealSense model that uses LiDAR technology, so I'm not sure if we can draw conclusions based on issues taken from the 400 series.

---

> I would recommend first trying to swap the USB cables of the two cameras in order to confirm whether the f0350132 camera has the same deviation in depth measurement

and

> You could also try unplugging the micro-sized end of the USB cable from the base of the f0270351 camera, turning it around the opposite way and re-inserting it

I tried swapping the cables, using different cables, and rotating the end of the cable that plugs to the camera. The outcome I have observed was the same regardless of the cable.

And to be clear, the depth of **f0350132** is not drifting/deviating with time. It is off by approximately 4--5 cm right from the moment the camera is plugged in.

MartyG-RealSense, 2022-10-03:
When electronic circuit boards in general become very hot, it can result in problems caused by short-circuits.  The recommended maximum operating temperatures listed in the data sheet document for the L515 camera model are quoted in the image below.

[image]

I note though your statement that the depth deviation occurs from the start of streaming and not over time, indicating that excessive temperature is likely not the cause of the issue.

I also note that the **Alternate IR** option is enabled on both cameras.  When enabling alternate IR, the IR image is holding the amplitude of the depth correlation.  This results in a noisy-looking infrared image.  As the Alternate IR option is disabled by default, are the two cameras more equal if you disable this option for both cameras in the Viewer?

ferrolho, 2022-10-03:
The table below shows the output of the IR and depth streams with the "Alternate IR" option _disabled_.

| f0270351                                                                                                                | f0350132                                                                                                                |
| ----------------------------------------------------------------------------------------------------------------------- | ----------------------------------------------------------------------------------------------------------------------- |
| [image]    | [image]    |
| [image] | [image] |

We can see that the IR of the **f0270351** (left) is much brighter and clearer than the IR of the **f0350132** (right). Further, notice how the depth stream of the camera on the left is more "complete" than the one on the right (which has greater areas without valid depth information).

MartyG-RealSense, 2022-10-04:
Given how the f0350132 camera's IR is darker, does changing the Visual Preset on that camera make a difference?

[image]

ferrolho, 2022-10-04:
It does not. I've set them to the same values, but the results are the same.

MartyG-RealSense, 2022-10-05:
Next, please try changing **Digital Gain** from the default 'High Gain' to 'Low Gain'.  This should darken the IR whilst leaving the depth image unaffected.

**High Gain**

[image]

**Low Gain**

[image]

MartyG-RealSense, 2022-10-12:
Hi @ferrolho  Do you require further assistance with this case, please?  Thanks!

ferrolho, 2022-10-12:
Hi @MartyG-RealSense, apologies for the delay in getting back to you.  I have tried different values for the gain (high/low), but it did not improve the output streams of the camera on the right-hand side of my analysis above (the **f0350132**).  I will try to return it and get a replacement for it.  Thank you for your help.

MartyG-RealSense, 2022-10-12:
You are very welcome, @ferrolho - thanks very much for the update.

closed 2022-10-12 by ferrolho

Edohvin, 2023-12-14:
@ferrolho Did you get a new camera? And did this solve the problem?

ferrolho, 2023-12-14:
I tried to return it and getting replacements, but the reseller never got back to me about the case. I've got a small graveyard of ~12 L515 on my desk. Anyway, I've started to use cameras from another manufacturer — but they don't come without their own issues either.

Edohvin, 2023-12-14:
I am using two D415's and I have different depth measurements from both cameras as well. The difference at a distance of 1.8m is about 6cm. I think I'll try a tare calibration later today. Let's see if that helps..

ferrolho, 2023-12-14:
Good luck! 🤞