realsenseai/librealsense #10963: Sensor stability

anshkumar, 2022-10-04:
Hi I was using intel real-sense D405 depth camera. I'm measuring a single object if size 8.8mm from an height of around 7.5 cm. All the imags were taken without disturbing the object or camera.

Here, are the results

	base_w	pred_w		diff
1	8.8	8.833169937		-0.033169937
2	8.8	8.696210571		0.103789429
3	8.8	8.811881766		-0.011881766
4	8.8	8.64700973		0.15299027
5	8.8	8.493011817		0.306988183
6	8.8	8.78972467		0.01027533
7	8.8	8.608452976		0.191547024
8	8.8	8.745411411		0.054588589
9	8.8	8.472187445		0.327812555
10	8.8	8.766700514		0.033299486

As one can see, there is a variation of 0.33mm.  Why so much fluctuations in the results.

I'm using the code from https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

Any idea how to reduce this variation ?

MartyG-RealSense, 2022-10-04:
Hi @anshkumar  Does the object that is being observed have a smooth or reflective surface with low texture detail or no texture?  Because the D405 camera model is not equipped with a built-in IR projector, it cannot cast a pattern of dots onto the surface of the object for the camera to use as a 'texture source' to analyze for depth information.  The camera may therefore have greater difficulty in retrieving depth detail from a smooth surface.

If the object has a smooth surface but is not reflective, casting a strong light source onto the surface or using an external *visible pattern projector** to cast onto the surface a pattern that the camera is able to see may help to improve depth analysis.

If you have fluorescent lights such as ceiling strip lights in the location that the camera is being used in then this could also be a source of noise, as such lights can flicker at a frequency that is difficult to see with the human eye.  Changing the **Power Line Frequency** option can help to negate this effect.  For North American regions, setting the Power Line Frequency to '60' is advisable, whilst European regions are likely to use '50'.

anshkumar, 2022-10-04:
Hi, 
Thanks for the reply. I'm using a textured surface object. I'm sharing both RGB and depth information in this [zip](https://drive.google.com/file/d/1PIoyuQNigBhR1P611XnN8ckh9PZ8VGnK/view?usp=sharing). I'm also sharing the base surface on which the object is placed [here](https://drive.google.com/file/d/1uCaqcac1Nq5-MUE0dFlvXh8_-ZBt96Qt/view?usp=sharing).

anshkumar, 2022-10-05:
Yes, I've already made those changes in my script.

MartyG-RealSense, 2022-10-05:
Looking at the measurements, the problem does not seem to be with measurement accuracy (which is excellent compared to the actual real-world 8.8 mm measurement of the observed object) but with the very small but frequent fluctuations in the depth value.

Please try adding a post-processing **temporal filter** to your Python script, and setting **Filter Smooth Alpha** to '0' (instead of the default '0.4').  If the depth reading is stable in the part of the image that the value is read from (i.e it is a solid color and not flickering with holes) then setting smooth alpha to zero should cause the depth value to stop fluctuating and remain constant.  

A downside of doing so is that the depth image takes longer to update if the camera is moved, but it sounds as though you are measuring the object whilst the camera is in a fixed position and so this update delay would not be a problem if the camera is stationary.

You can test the temporal filter setting first in the RealSense Viewer tool to see whether it stops fluctuation.  If you enable the filter with the '0' setting for alpha and then move the mouse cursor over the depth image then you can see the depth reading in the bottom corner of the Viewer's depth stream panel.

[image]

A simple example of a script for implementing a temporal filter in Python and setting the smooth alpha would be something like this:

```
import pyrealsense2 as rs
pipe = rs.pipeline()
cfg = rs.config()
cfg.enable_stream(rs.stream.Depth, 640, 480, rs.format.Z16, 30);
pipe.start(cfg)

# Declare filter
temp_filter = rs.temporal_filter(smooth_alpha=0, smooth_delta = 80, persistence_control=8)

# Apply filter

filtered = temp_filter.process(filtered)
```

anshkumar, 2022-10-05:
Will try it. I've also observed that measuring the height gives more error than the width. Above were results when width of the object was measured. Here are the results of height of the object measured (Taking difference of the minimum of the object height form the flat surface):

```
height pre | height acc | h diff
7.3 | 6.7 | 0.6
7.2 | 6.7 | 0.5
7.8 | 7.6 | 0.2
8.4 | 8.4 | 0
8 | 7.83 | 0.17
7.7 | 7.6 | 0.1
9.1 | 8.5 | 0.6
7.8 | 7.57 | 0.23
8.2 | 8 | 0.2
7.9 | 7.3 | 0.6
9 | 8.9 | 0.1
8.7 | 8 | 0.7
7.6 | 7.2 | 0.4
9.2 | 8.25 | 0.95
8.7 | 8 | 0.7
```
A difference of 0.95 mm was found. Currently running in default mode. Any suggestions?

MartyG-RealSense, 2022-10-05:
Looking at your image of the object, I note that it has a shadow around it at the top and bottom of its height.  There was a previous case at https://github.com/IntelRealSense/librealsense/issues/7021 in which a box was being observed from above and the shadow from the box was creating measurement error.  A similar phenomenon may be occurring here.

[image]

anshkumar, 2022-10-06:
Actually for us that's the width of the object. The height of the object is along the z axis. In the image below it will be the distance of the black dot from the base.
[image]

MartyG-RealSense, 2022-10-06:
In the coordinate system of RealSense cameras, the Z axis represents the 3D distance of the object from the camera, whilst X and Y represent width and height.  So Z could be described in terms of the distance in meters to the center of the top of the object shown by the black dot, minus the distance in meters from the camera to the background beneath the object (which is a known value if the camera is fixed) 

Would it be possible to temporarily place a different surface under the object to confirm or eliminate the possibility that the backdrop of the material on the rollers is difficult for the camera to read?

anshkumar, 2022-10-10:
The error is mainly due to the angle at which the measurement is made. I also noticed one thing. In my observation I found that one pixel was equivalent to 0.2 mm at 7.5 to 8 cm distance from the object. However I was expecting it to be 0.1 mm since the camera accuracy is 0.1 mm at 7cm distance. Are you also having same mm to pixel value ?

MartyG-RealSense, 2022-10-11:
The default depth unit scale of the D405 is **0.01**, whilst all of the other 400 Series models have a default of 0.001.  The officially stated depth accuracy is **< 2% at 50 cm**.

https://www.intelrealsense.com/compare-depth-cameras/

anshkumar, 2022-10-11:
Ok. I was referring to the [article](https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#move-as-close-as-possible). It says that by changing the 'Disparity shift', I can reduce MaxZ and minZ. What's the equivalent of it in python? I'm not able to find in the examples. And also how to change the depth setting to 'High Accuracy' in python ?

MartyG-RealSense, 2022-10-11:
Reducing the Disparity Shift value to try to lower the minimum depth sensing distance will not make a difference on the D405 model, as advised at https://github.com/IntelRealSense/librealsense/issues/10444#issuecomment-1108842789

https://github.com/IntelRealSense/librealsense/issues/2015#issuecomment-403410366 has an example of code for setting the Disparity Shift value in Python.

anshkumar, 2022-10-11:
I tried Disparity Shift and I was able to reduce the height from 8 cm to 4 cm.
Here, is the result with Disparity Shift of 100:
[image]

and with Disparity Shift of 0:
[image]

Also, I made Disparity Mode from zero to one.

MartyG-RealSense, 2022-10-11:
That's very interesting that you were able to reduce minimum distance on your D405!  I had previously tested with SDK 2.50.0 at https://github.com/IntelRealSense/librealsense/issues/10444#issuecomment-1108842789 and whilst it could work with D405, official support for this model was not introduced until 2.51.1 which you are using.  I will update my comments to reflect this.  Thanks very much!

MartyG-RealSense, 2022-10-16:
Hi @anshkumar  Do you require further assistance with this case, please?  Thanks!

anshkumar, 2022-10-16:
No, that's all.

closed 2022-10-16 by anshkumar