realsenseai/librealsense #10964: Introducing the RealSense D457 GMSL/FAKRA stereo camera

MartyG-RealSense, 2022-10-04:
Hi everyone,

Intel have introduced a new member of the RealSense stereo camera family called the **D457**.  It is based around the D450 depth module and the new **Vision Processor D4 V5**. 

https://www.intelrealsense.com/depth-camera-d457/

[image]

Orders for the D457 can be placed now on the official RealSense Store.

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d457.html

To quote the store page:

"The Intel® RealSense™ Depth Camera D457 is our first GMSL/FAKRA high bandwidth stereo camera.  The D457 has an IP65 grade enclosure protecting it from dust ingress and projected water.  D457 has an on-board Maxim serializer chip.  A Maxim de-serializer is needed at host side to decode the streams".

*********

The dedicated data sheet document for the D457 can be downloaded here:

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

closed 2022-10-09 by MartyG-RealSense

dennishuneou, 2023-03-15:
Can i use only two lanes for D457 instead of three lanes?
Since only two lanes wired on our boards.
Thank you.

MartyG-RealSense, 2023-03-15:
Hi @dennishuneou  Are you referring to the MIPI CSI-2 lanes, please?  If you are, the table below from the data sheet document for the D457 camera lists 2 lanes for the interface to the Vision Processor D4 and 1 lane for the interface to the RGB sensor.

[image]

dennishuneou, 2023-03-15:
[image]
when using GMSL2 for transmission,
Depth stream used one lane.
Left Image[y8] and IMU used one lane.
RGB used one lane.
Can i use only 2 lanes for 4 streams.

MartyG-RealSense, 2023-03-15:
My colleagues confirmed that 2 lanes are being used now on D457.

dennishuneou, 2023-03-20:
Is this mandatory?
I’m working for D457 on our hardware platform.
### HW-SYNC  ###
# SerDes Depth Trigger Path MFP7 > MFP8
sudo i2cset -f -y 2 0x48 0x02 0x82c5 w #MFP7
sudo i2cset -f -y 2 0x48 0x02 0x1fc6 w
sudo i2cset -f -y 2 0x40 0x02 0x84d6 w #MFP8
sudo i2cset -f -y 2 0x40 0x02 0x60d7 w #OUT_TYPE bit to 1
sudo i2cset -f -y 2 0x40 0x02 0x1fd8 w

# SerDes RGB Trigger Path MFP9 > MFP0
sudo i2cset -f -y 2 0x48 0x02 0x82cb w #MFP9
sudo i2cset -f -y 2 0x48 0x02 0x1bcc w
sudo i2cset -f -y 2 0x40 0x02 0x84be w #MFP0
sudo i2cset -f -y 2 0x40 0x02 0x60bf w #OUT_TYPE bit to 1
sudo i2cset -f -y 2 0x40 0x02 0x1bc0 w

Best Regards,
Dennis Hu

Neousys Technology America Inc.
3384 Commercial Ave., Northbrook, IL 60062 | (D) 847-656-3395 | (P) 847-656-3298 X407

MartyG-RealSense, 2023-03-20:
Hi @dennishuneou  Could you provide further information about your question, please?  Are you asking if D457 having 2 lanes instead of 3 is mandatory?

dennishuneou, 2023-03-20:
Hi ,

I ported d4xx driver and dts into Xavier NX.
I can’t capture video when running realsense-viewer.
I read out registers of Max9295, I found that the pixclk is not locked.
So, I think that maybe we must send external trigger signal to camera.
The following content comes from Serdes_D457_1.0.0.A.sh
### HW-SYNC  ###
# SerDes Depth Trigger Path MFP7 > MFP8
sudo i2cset -f -y 2 0x48 0x02 0x82c5 w #MFP7
sudo i2cset -f -y 2 0x48 0x02 0x1fc6 w
sudo i2cset -f -y 2 0x40 0x02 0x84d6 w #MFP8
sudo i2cset -f -y 2 0x40 0x02 0x60d7 w #OUT_TYPE bit to 1
sudo i2cset -f -y 2 0x40 0x02 0x1fd8 w

# SerDes RGB Trigger Path MFP9 > MFP0
sudo i2cset -f -y 2 0x48 0x02 0x82cb w #MFP9
sudo i2cset -f -y 2 0x48 0x02 0x1bcc w
sudo i2cset -f -y 2 0x40 0x02 0x84be w #MFP0
sudo i2cset -f -y 2 0x40 0x02 0x60bf w #OUT_TYPE bit to 1
sudo i2cset -f -y 2 0x40 0x02 0x1bc0 w

Best Regards,
Dennis Hu

Neousys Technology America Inc.
3384 Commercial Ave., Northbrook, IL 60062 | (D) 847-656-3395 | (P) 847-656-3298 X407

MartyG-RealSense, 2023-03-20:
Camera firmware driver 5.14.0.0 and RealSense SDK version 2.53.1 are minimum requirements.  Can you confirm if you are using this firmware and SDK version please?

The MIPI reference driver is also designed for use with Intel's own de-serializer board at the link below.  Furthermore, the MIPI driver is designed for use with AGX Xavier specifically, whilst you are using Xavier NX.  I note though that you have ported the driver, which should be a solution for these issues.

https://store.intelrealsense.com/buy-intel-realsense-des457.html

dennishuneou, 2023-03-20:
I bought camera week ago.
The firmware version is 5.13.0.53.
Realsense SDK is 2.50.0. installed from vcpkg.

Best Regards,
Dennis Hu

Neousys Technology America Inc.
3384 Commercial Ave., Northbrook, IL 60062 | (D) 847-656-3395 | (P) 847-656-3298 X407

https://store.intelrealsense.com/buy-intel-realsense-des457.html

MartyG-RealSense, 2023-03-20:
SDK 2.50.0 does not have MIPI support as it had not been implemented yet in that SDK version.  SDK 2.53.1 which introduces MIPI support should be used, and firmware should be updated to 5.14.0.0.

If MIPI is not an absolute requirement for your project then there is also the option to move a switch under the camera's removable casing cap from 'MIPI' position to 'USB' position so that it is detected and performs as a D455 USB camera, as described at https://github.com/IntelRealSense/librealsense/issues/11010#issuecomment-1433488437