realsenseai/librealsense #10966: How to obtain infrared data of L515?

liyuan1a, 2022-10-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.50.0 }                          | 
| Language                            |  {matlab }                          | 
| Segment			|  {others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

The official documentation provides some typical sample programs of matlab. Through the example program, I know how to obtain depth data. I want to ask how to obtain infrared data.

pointcloud = realsense.pointcloud();
fs = pipe.wait_for_frames();
depth = fs.get_depth_frame();
points = pointcloud.calculate(depth);
infr=fs.get_infrared_frame();

We can obtain point cloud value points through "pointcloud.calculate(depth)", but how do we obtain the value of "infr"?

MartyG-RealSense, 2022-10-05:
Hi @liyuan1a  Are you able to obtain the infrared frames if you insert an index number '0' into the brackets of the infr line?

**infr=fs.get_infrared_frame(0);**

liyuan1a, 2022-10-05:
Whether the quotation mark is 0 in the bracket or not, we can get the data in "1x1 video_frame" format, but I can't get the specific value in this data format.

infr=fs.get_infrared_frame(0);
infr2=fs.get_infrared_frame();
[image]

MartyG-RealSense, 2022-10-05:
As you are defining depth and infrared and not depth and color, is your goal to map infrared coordinates to depth coordinates on a pointcloud instead of mapping color onto the pointcloud?  If you are aiming for a pointcloud that has infrared mapped to depth, would the code below work for your project?

```
depth = fs.get_depth_frame();
infr=fs.get_infrared_frame();
pointcloud.map_to(infr);
points = pointcloud.calculate(depth);
```

liyuan1a, 2022-10-06:
I'm sorry, maybe I didn't express my meaning clearly. My main purpose is to know the infrared value in this cloud.
[image]
As shown in the picture, I want to know how to get the value "0X8e" through matlab code.

MartyG-RealSense, 2022-10-06:
There is not code available for retrieving the XY coordinates of the infrared stream specifically.  This is likely because in most circumstances it is not necessary.  Because infrared and depth originate from the same depth sensor, separate depth and infrared images will have the same XY coordinate for the same pixel area on each image.  So you could obtain 2D infrared XY by obtaining the 2D depth XY coordinate.

[image]

closed 2022-10-18 by MartyG-RealSense