realsenseai/librealsense #10967: D405 crashes app which runs smoothly on SR300 and D435

nickelnext, 2022-10-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 128223070567 | 
| Operating System & Version |  Win 11  | 
| Platform                                 | PC |
| SDK Version                            | 2.52 }                          | 
| Language                            |C#                          | 

### Issue Description
Hi @MartyG-RealSense 

Do you have any idea if there is some specific instruction/functions that we need to use for D405 for accessing RGB/Depth streams?

We have a reconstruction algorithm which runs smoothly with sr300 and d435, but when used with D405 the app crashes with no explanation (we are calling a C++ dll and in our C# code we have marhaling methods which run fine on those two cameras).

I believe it's either the configuration of the color/depth width/height, or maybe the rs2 methods to access the rgb camera, which do make crash our app.

In the sdk code it looks that in order to support d405 the developers have inserted some specific functions, but it's not clearly stated whether we should update also the way how we access the rs2 APIs or not

Thank you

MartyG-RealSense, 2022-10-05:
Hi @nickelnext  I note that you list your SDK version as 2.52.  If you are using 2.52, that version is a **pre-release** that is designed for the RealSense SDK's new MIPI and GMSL support - see https://github.com/IntelRealSense/librealsense/issues/10964  

For your project you will likely need to be using the dll from the **2.51** SDK version.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

nickelnext, 2022-10-06:
Hi Marty

For Sr300 we're using 640 x 480 on depth and 1920x1080 for color
For D435 we're using 1280x800 on depth and 1920x1080 for color
For D405 we tried several resolutions but none is working.

MartyG-RealSense, 2022-10-06:
The resolutions supported on the D405 model are listed below.  The D405 does not have a separate RGB sensor and instead obtains RGB from the depth sensor, so 1920x1080 color is not supported. 

[image]

1288x808 is only available for Y16 infrared.  For both depth and color, the maximum supported resolution is **1280x720**

[image]

MartyG-RealSense, 2022-10-06:
The configuration looks okay, though I would recommend testing it by hard-coding the resolution values instead of substituting them in with width and height variables in order to confirm whether there is a problem with the structure of the `this->mColorImage.width, this->mColorImage.height `variables.

Also, have two separate definitions for depth and color streams if you do not already.

**rs2.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
rs2.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGRA8, 30);**

nickelnext, 2022-10-07:
Yep our code looks like
`rs2::config cfg;`
			`cfg.enable_stream(RS2_STREAM_COLOR, this->mColorImage.width, this->mColorImage.height, RS2_FORMAT_BGRA8, 30);`
			`cfg.enable_stream(RS2_STREAM_DEPTH, this->mDepthImage.width, this->mDepthImage.height, RS2_FORMAT_Z16  , 30);`

`rs2::pipeline pipe;`
`rs2::pipeline_profile profile = pipe.start(cfg);`

We also tried to hardcode them but it doesn't change the fact that we get an accessViolation exception, both with 2.51 and 2.52.

The same code runs with the Sr300 and d435

MartyG-RealSense, 2022-10-07:
Is 1280x720 for depth and color accessible on the D405 for the RealSense Viewer or do you get an access violation with the Viewer too?

closed 2022-10-20 by MartyG-RealSense