realsenseai/librealsense #10976: How can I change the depth unit of recording?

yumonorui, 2022-10-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   {Win 10   | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  {  2.51.1 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I am looking to change the depth units of my previous recording. In other words, I would like to recognize objects at 0.0001m instead of 0.001m. Is it possible to do that? Or do I need to change the depth unit in the contorol section of the viewer when recording? When I changed the depth unit, it did not change properly.
Also, if I shoot at a very close distance, it recognized the 0.0001m object as mentioned above, but there were many unmeasurable areas, which is not the result I am looking for.

MartyG-RealSense, 2022-10-07:
Hi @yumonorui  The ideal depth range of the D405 camera model is 7 cm to 50 cm.  If the camera is placed closer than 7 cm to an observed object then the depth image will begin progressively breaking up as the camera is moved closer to the object.

The default depth unit scale of the D405 is **0.01**, whilst the other 400 Series models use the default of '0.001'.  At '0.0001' you will therefore lose most of the depth detail.  '0.001' may therefore work better for you.

If you are using a bag file as the data source instead of a live camera then the SDK should treat the bag file as though it is live camera data.  You could set the depth units in Python like this:

**depth_sensor.set_option(rs.option.depth_units,0.001)**

This is demonstrated in a Python script at https://github.com/IntelRealSense/librealsense/issues/2870

[image]

yumonorui, 2022-10-11:
Thank you for your answering and I'm sorry for the late response. I try this way but I got another problem:
` depth_sensor.set_option(rs.option.depth_units,0.001)
RuntimeError: This option is read-only!`
I searched it but I can't get the helpful information. Thanks.

MartyG-RealSense, 2022-10-12:
Possibly it is because a bag file is being used and so there is not a real camera depth sensor present to change the depth scale of.  

There is another approach that you could try to improve your image.  A D405 user at https://github.com/IntelRealSense/librealsense/issues/10963#issuecomment-1274941411 found that now that the D405 is officially supported in SDK version 2.51.1 onwards, the **Disparity Shift** option for reducing the minimum distance of the camera now worked, whereas in the pre-official 2.50.0 version the option did not make a difference.

An example of Python code for setting the Disparity Shift option can be found at https://github.com/IntelRealSense/librealsense/issues/2015#issuecomment-403410366

closed 2022-10-24 by MartyG-RealSense