realsenseai/librealsense #10978: Realsense camera bag file : Concatenation

PierreLouisT, 2022-10-07:
Hello ! 
I have made a lot of bag file thanks to the code : read_bag_from_example. 
I have done those with one single frame so the video is less than 1 second, and my objective is to concatenate all of those bag files and get all depth data, color data and IR data. I can do it with only one bag as I said. 
I think I can use this part of the program from read_bag_from_example

      `parser = argparse.ArgumentParser(description=" ")
        # Add argument which takes path to a bag file as an input
       parser.add_argument("-i", "--input", type=str, help="")
        # Parse the command line arguments to an object
        args = parser.parse_args()
        # Safety if no parameter have been given
        if not args.input:
            print("No input paramater have been given.")
            print("For help type --help")
            exit()
        # Check if the given file have bag extension
        if os.path.splitext(args.input)[1] != ".bag":
            print("The given file is not of correct file format.")
            print("Only .bag files are accepted")
            exit()    
        self.pc = rs.pointcloud()
        self.pipeline = rs.pipeline()
        config = rs.config() `

Maybe I can add manually all my bag file ? 
Is someone experienced this case ? 
Thanks in advance, 

PL

MartyG-RealSense, 2022-10-07:
Hi @PierreLouisT  It is possible to read a set of multiple bag files by using a **loop**. 
 https://github.com/IntelRealSense/librealsense/issues/6887 may be a helpful Python reference for doing so.  This case also has a link to a C++ example of reading multiple bag files in a loop at https://github.com/IntelRealSense/librealsense/issues/2693

PierreLouisT, 2022-10-11:
Hello Marty ! 
Thank you for your help, I got solution I needed, 
I can close the issue

closed 2022-10-11 by PierreLouisT

MartyG-RealSense, 2022-10-11:
It's great to hear that you were successful - thanks very much for the update!