realsenseai/librealsense #10980: Certificate verification failed: The certificate is NOT trusted for distribution_linux.md

huangjiafengx, 2022-10-08:
Hello!

I install Intel® RealSense ™ SDK on the ubuntu, find the Certificate verification failed by the step:
sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u

[image]

Thanks in advance.

FeiSong123, 2022-10-08:
I also encountered this problem ！

MartyG-RealSense, 2022-10-08:
I will refer this certificate issue to my Intel RealSense colleagues.  Thanks very much @huangjiafengx and @FeiSong123 for the reports!

erikkocky, 2022-10-08:
Same here 
Ubuntu 20.04, 5.15.0-48 kernel

MartyG-RealSense, 2022-10-08:
Thanks very much for adding your report, @erikkocky 

I would add that whilst your kernel version is likely unrelated to the certificate issue, kernel 5.15 is not currently officially supported by the librealsense SDK.  Unsupported kernels can be used with the SDK but there may be unpredictable consequences in regards to stability.  An unofficial test patch that adds kernel 5.13 and 5.15 support can be found at https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1171965106

erikkocky, 2022-10-08:
Hey @MartyG-RealSense , Yes I am aware of that thanks, but for now everything was working fine for me. 
When will you release update for newer kernels ?

MartyG-RealSense, 2022-10-08:
I have also passed to my colleagues the question about future support for newer kernels on package installations.

Kernel support updates for source code installations of librealsense are added over time, and there is also the option to build librealsense from source code with the **RSUSB backend** installation method that bypasses the kernel and so is not dependent on Linux versions or kernel versions and does not require patching.

clogwog, 2022-10-09:
any update on the https://librealsense.intel.com certificate being expired ?

MartyG-RealSense, 2022-10-10:
Hi @clogwog  My Intel RealSense colleagues are aware of the certificate issue and are looking into it

dburr, 2022-10-10:
@huangjiafengx @FeiSong123 @erikkocky @clogwog and others still experiencing this, here is a workaround. Not recommended, but if you need the RS packages, this is something to consider. You can disable Apt's certificate checking by issuing the following commands:

```
touch /etc/apt/apt.conf.d/99verify-peer.conf \
&& echo >>/etc/apt/apt.conf.d/99verify-peer.conf "Acquire { https::Verify-Peer false }"
```

The security implications of doing this should be obvious, so I wouldn't recommend leaving this in permanently. Once you successfully install the Realsense deb, you can undo this step by removing the file (`rm /etc/apt/apt.conf.d/99verify-peer.conf`). (you'll get "certificate not trusted" errors whenever you `apt update` but it won't affect the installed Realsense deb.

MartyG-RealSense, 2022-10-10:
Thanks very much @dburr for your advice!

clogwog, 2022-10-10:
@dburr thank you ! I only need to do an apt-update to i've temporarily changed it to 
`apt-get -o Acquire::https::Verify-Peer=false update`

bmegli, 2022-10-10:
Ideally ignore check only for Intel.

You will also need to ignore for install:

```bash
# Workaround Intel certificate expired on 10.10.2022
sudo apt -o Acquire::https::librealsense.intel.com::Verify-Peer=false update
sudo apt -o Acquire::https::librealsense.intel.com::Verify-Peer=false install librealsense2 librealsense2-net librealsense2-utils librealsense2-gl
```

andresgalaviz, 2022-10-10:
Combining the great suggestions of @bmegli and @dburr you can do this, tested for both update and install:

```bash
sudo sh -c "echo 'Acquire::https::librealsense.intel.com::Verify-Peer \"false\";' > /etc/apt/apt.conf.d/99verify-peer.conf"
```

This file should be removed after this is fixed.

gkrasin, 2022-10-10:
Here is a trick that still does certificate checking, but acknowledges the fact that Intel let it to expire. TL,DR: temporarily set the date to Oct 05, 2022:

```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
# Turn off NTP sync
sudo timedatectl set-ntp false
sudo date --set '2022-10-05 12:34:56'
sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u
sudo apt-get install librealsense2-dkms librealsense2-utils librealsense2-dev
# Turn on NTP sync, the time jumps to the real one
sudo timedatectl set-ntp true
```

jwnimmer-tri, 2022-10-10:
Changing the date _globally_ on the machine will probably do lots of awful things and explode a bunch of other unrelated software.  I strongly advise against it!

gkrasin, 2022-10-10:
@jwnimmer-tri depends on what's running on the machine. But yes, to be used with caution.

MartyG-RealSense, 2022-10-11:
@erikkocky  Regarding your earlier question at https://github.com/IntelRealSense/librealsense/issues/10980#issuecomment-1272300866 about future support for kernels newer than 5.4 when installing from packages, my Intel RealSense colleagues say that it is being discussed.

micuentadecasa, 2022-10-11:
> Combining the great suggestions of @bmegli and @dburr you can do this, tested for both update and install:
> 
> ```shell
> sudo sh -c "echo 'Acquire::https::librealsense.intel.com::Verify-Peer \"false\";' > /etc/apt/apt.conf.d/99verify-peer.conf"
> ```
> 
> This file should be removed after this is fixed.

This worked for me in WSL2

andresgalaviz, 2022-10-11:
> Combining the great suggestions of @bmegli and @dburr you can do this, tested for both update and install:
> 
> ```shell
> sudo sh -c "echo 'Acquire::https::librealsense.intel.com::Verify-Peer \"false\";' > /etc/apt/apt.conf.d/99verify-peer.conf"
> ```
> 
> This file should be removed after this is fixed.

The issue appears to be resolved now, if you used this don't forget to remove the file!
`sudo rm -f /etc/apt/apt.conf.d/99verify-peer.conf`

sahilshembekar, 2022-10-12:
Has this issue been resolved?

ferrolho, 2022-10-12:
> Has this issue been resolved?

It seems so. At least on my side, no problems today when doing `sudo apt update`. 

[image]

huangjiafengx, 2022-10-12:
> > Has this issue been resolved?
> 
> It seems so. At least on my side, no problems today when doing `sudo apt update`.
> 
> [image]

Thank you!   it is ok now!

closed 2022-10-12 by huangjiafengx

MartyG-RealSense, 2022-10-12:
Thanks very much everyone for your confirmations of the correction of the certificate issue  :)

hernot, 2022-10-12:
No errors here anymore too. Latest version reported for ubuntu 20.04 (mint 20) by apt-cache is 2.51.1-0~realsense0.7526.