realsenseai/librealsense #10983: apt update repository https://librealsense.intel.com/Debian/apt-repo focal Release does not contain Release-file any more

hernot, 2022-10-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|sudo add-apt-repository "deb https://librealsense.intel.com/Debia------------------------------------------- |
| Camera Model                       | Not relevant | 
| Firmware Version                   | Not relevant |
| Operating System & Version |    Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    | Not relevant                                         | 
| Platform                                 | PC  | 
| SDK Version                            |  Trying to update from 2.51.1                         | 
| Language                            |  Not relevant                          | 
| Segment			|  Not relevant                         | 

### Issue Description
When running apt-get update on ubuntu 20.04 i get the Follwing error (translated from German)
The repository Depot »https://librealsense.intel.com/Debian/apt-repo focal Release« does not contain the Release file any more. HAs the support ended or just was that file accidentally not transferred to the repository by the last release.

MartyG-RealSense, 2022-10-09:
Hi @hernot  My Intel RealSense colleagues are aware of the certificate issue and are looking into it.  Thanks very much for the report and for your patience!

hernot, 2022-10-10:
@MartyG-RealSense thanks you are right in the end it boils down to being duplicate of #10980. Would you mind marking this issue duplicate of #10980 to ensure it is closed along with it when solved.

MartyG-RealSense, 2022-10-10:
It is no problem to keep this case open until https://github.com/IntelRealSense/librealsense/issues/10980 is resolved and close this case when that one is closed.  Thanks!

MartyG-RealSense, 2022-10-10:
Thanks very much @130s for adding your own report to the RealSense users who are experiencing this issue.

ripdk12, 2022-10-12:
This problem is solved for me now. I'd bet it's safe to close this issue now.

closed 2022-10-12 by hernot