realsenseai/librealsense #10985: L515, Receiver light saturation error

TheNemo05, 2022-10-10:
*
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Ubuntu 20  | 
| Kernel Version (Linux Only)    | 5.14.0-1052-oem |                                        | 
| Platform                                 | PC |
| SDK Version                            |  2.51.1 | 
| Language                            |  ROS | , Python

### Issue Description

prasad@prasad:~$ roslaunch realsense2_camera demo_pointcloud.launch 
... logging to /home/prasad/.ros/log/9384b62c-485a-11ed-8f25-db5a6b863d8e/roslaunch-prasad-3272.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://prasad:46369/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 0
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
[... 81 lines cut ...]
[ INFO] [1665378953.344606848]: Device Serial No: f0246007
[ INFO] [1665378953.344617202]: Device physical port: 3-1-2
[ INFO] [1665378953.344626958]: Device FW version: 01.05.08.01
[ INFO] [1665378953.344638063]: Device Product ID: 0x0B64
[ INFO] [1665378953.344647258]: Enable PointCloud: On
[ INFO] [1665378953.344655563]: Align Depth: Off
[ INFO] [1665378953.344666389]: Sync Mode: On
[ INFO] [1665378953.344730508]: Device Sensors: 
[ INFO] [1665378953.441244369]: L500 Depth Sensor was found.
[ INFO] [1665378953.461009292]: RGB Camera was found.
[ INFO] [1665378953.461485857]: Motion Module was found.
[ INFO] [1665378953.461626426]: Add Filter: pointcloud
[ INFO] [1665378953.462718183]: num_filters: 1
[ INFO] [1665378953.462772286]: Setting Dynamic reconfig parameters.
[ INFO] [1665378954.824638530]: Done Setting Dynamic reconfig parameters.
[ WARN] [1665378954.824839051]: Given stream configuration is not supported by the device!  Stream: Depth, Stream Index: 0, Width: 640, Height: 480, FPS: 30, Format: Z16
[ WARN] [1665378954.824860994]: Using default profile instead.
[ INFO] [1665378954.827225521]: depth stream is enabled - width: 320, height: 240, fps: 30, Format: Z16
[ INFO] [1665378954.827632064]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ WARN] [1665378954.827654572]: Given stream configuration is not supported by the device!  Stream: Confidence, Stream Index: 0, Width: 640, Height: 480, FPS: 30, Format: None
[ WARN] [1665378958.828362891]: Hardware Notification:Receiver light saturation, stream stopped for 1 sec,1.66538e+12,Error,Hardware Error

A warning displayed as Receiver light saturation error, which is causing flicking of 3D image in RVIZ pointcloud. Any solutions would be appreciated. 

@RealSenseSupport , @MartyG-RealSense

TheNemo05, 2022-10-10:
The camera is already out of warranty & has no scope of return.

MartyG-RealSense, 2022-10-10:
As this error is related to a problem in the L515 camera's hardware, it is unfortunately not correctable with software.

MartyG-RealSense, 2022-10-16:
Hi @TheNemo05  Bearing in mind the above comment, do you require further assistance with this case, please?  Thanks!

closed 2022-10-25 by MartyG-RealSense