realsenseai/librealsense #10987: No Frames Received!

liyuan1a, 2022-10-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515} | 
| Firmware Version                   | (Open RealSense Viewer --> 2.50.0) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Why does "No Frames Received!"  appear after the software Intel RealSence Viewer runs for a period of time?

[image]
[image]

MartyG-RealSense, 2022-10-11:
Hi @liyuan1a  The "No frames received!" warning means that new images (frames) are no longer being received.  It looks as though one particular camera in your set of multiple L515s is being affected whilst the others are working normally. 

On the affected camera, could you first try unplugging the micro-sized end of the USB cable from the base of the camera, reversing the connector's orientation and plugging it back into the L515 camera, please (USB-C cables are two-way insertion at the micro-sized end).  Does doing so make any difference?

liyuan1a, 2022-10-12:
We have tried this. After restarting the camera L515, we can receive frames, but after a period of time (about 30s, 1 min or 5 min), some cameras still  suddenly appear "No frames received!". It seems that the L515 camera is unstable, which has a great impact on our measurement. How to keep the camera stable during operation？
  In addition, when we run, sometimes some cameras are disconnected. We need to restart the computer to connect the camera normally again.

closed 2022-10-24 by MartyG-RealSense