realsenseai/librealsense #10988: Build and apply patched kernel modules fail on kernel 5.11

DeboBurro, 2022-10-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i| 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |  Ubuntu 20.04.1 | 
| Kernel Version (Linux Only)    | 5.11.0-46-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  built from [source](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)                 | 
| Language                            |  c, c++                          | 
| Segment			| Robot                   | 

### Issue Description
I followed the [instruction](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) to build librealsense2 and patch kernel modules.
Steps:
1. Remove all realsense cameras from the robot
2. `sudo apt remove librealsense2*`
3. `cd /tmp/new_librealsense_test/`
4. `git clone https://github.com/IntelRealSense/librealsense.git`
5. `cd /tmp/new_librealsense_test/librealsense `
6. Start to build the librealsense2 SDK from source
7. `mkdir build && cd build`
8. `cmake ../ -DCMAKE_BUILD_TYPE=Release`
9. `sudo make uninstall && make clean && make && sudo make install`
10. The librealsense2 SDK is installed properly from above, later I tried to build and install the patched kernel modules
11. `sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev`
12. `cd /tmp/new_librealsense_test//librealsense`
13. `./scripts/setup_udev_rules.sh`
14. `./scripts/patch-realsense-ubuntu-lts.sh`
15. The command above built the kernel modules but wasn't able to insert. The errors shown below
```
                  .
                  .
  LD [M]  /tmp/new_librealsense_test/librealsense/ubuntu-focal-hwe-5.11/drivers/media/v4l2-core/videodev.ko
  LD [M]  /tmp/new_librealsense_test/librealsense/ubuntu-focal-hwe-5.11/drivers/media/v4l2-core/videobuf-vmalloc.ko
make: Leaving directory '/tmp/new_librealsense_test/librealsense/ubuntu-focal-hwe-5.11'
Patched kernels modules were created successfully

Replacing videodev  -
        Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Exec format error
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  videodev  module was reloaded
```

MartyG-RealSense, 2022-10-12:
Hi @DeboBurro  In the past issues https://github.com/IntelRealSense/librealsense/issues/1225 and https://github.com/IntelRealSense/librealsense/issues/5598 where this problem was reported, the solution was found to be to disable **Secure Boot** in the computer's BIOS configuration.

DeboBurro, 2022-10-12:
@MartyG-RealSense Thank you. The secure boot was already be disabled in the beginning of provisioning the robots (nvidia-driver required us to disable the secure boot). I also double checked again now, so this might not be the case.

MartyG-RealSense, 2022-10-13:
I note in your installation list above that you are building librealsense (step 6) and then applying the kernel patch script (step 14).  The installation order described in the **installation.md** instructions is to apply the patch before building librealsense.

In regard to removing librealsense packages with **sudo apt remove librealsense2** in step 2 before performing a source code installation, the SDK's package installation instructions at **distribution_linux.md** recommend using a 'purge' instruction instead to remove all librealsense-related packages that might be installed on a computer.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

The installation_md instructions also advise to perform` sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev` before installing librealsense.

DeboBurro, 2022-10-13:
@MartyG-RealSense I followed your suggestions by the steps below
1. `sudo apt install ${librealsense_pkgs[@]}`  # install back the librealsense packages so later when it purges, it would remove all the config files properly
2. `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`
3. `sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev`
4. `mkdir -p /tmp/1013_realsense_test/ && cd /tmp/1013_realsense_test/ `
5. `git clone https://github.com/IntelRealSense/librealsense.git`
6. `cd librealsense`
7. `./scripts/patch-realsense-ubuntu-lts.sh`
8. But it still shows the same error
```
  LD [M]  /tmp/1013_realsense_test/librealsense/ubuntu-focal-hwe-5.11/drivers/media/v4l2-core/videobuf-core.ko
make: Leaving directory '/tmp/1013_realsense_test/librealsense/ubuntu-focal-hwe-5.11'
Patched kernels modules were created successfully

Replacing videodev  -
	Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Exec format error
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  videodev  module was reloaded
```
And I checked the kernel log by `dmesg -T`, the time when the kernel module inserts i
```
[Thu Oct 13 11:01:13 2022] USB Video Class driver (1.1.1)
[Thu Oct 13 12:02:05 2022] usbcore: deregistering interface driver uvcvideo
[Thu Oct 13 12:02:05 2022] mc: Linux media interface: v0.10
[Thu Oct 13 12:02:05 2022] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 00000000022915a0, val ffffffffc3972e7d
[Thu Oct 13 12:02:05 2022] videodev: Linux video capture interface: v2.00
```
I then removed all the linux header files and reinstall back the 5.11.0-46 header
```
sudo apt remove --purge linux-headers-*
sudo apt autoremove && sudo apt autoclean
sudo apt install linux-headers-5.11.0-46-generic
sudo reboot
```
Then double checked the system after rebooting
```
USERNAME@burro-8-883:~$ mokutil --sb-state
SecureBoot disabled
burro@burro-8-883:~$ uname -a
Linux burro-8-883 5.11.0-46-generic #51~20.04.1-Ubuntu SMP Fri Jan 7 06:51:40 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```
And follow the same steps in the first section to install but it still shows the `could not insert 'videodev': Exec format error`

MartyG-RealSense, 2022-10-14:
Is it possible for you to build librealsense with the **RSUSB backend**, which bypasses the Linux kernel and so provides hardware metadata without having to apply a kernel patch?  You can build librealsense in RSUSB mode by adding **-DFORCE_RSUSB_BACKEND=true** to the CMake build instruction.

`cmake ../ -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=true`

fiorano10, 2022-10-14:
@MartyG-RealSense we are using a multi camera set up and need to use the V4L backend

fiorano10, 2022-10-14:
[image]

MartyG-RealSense, 2022-10-14:
Hi @fiorano10  The comment about the ROS wrapper installation being based on the RSUSB backend relates to the **Method 1** installation method in the wrapper instructions that installs librealsense and the ROS wrapper together at the same time from packages with apt-get install, as advised at https://github.com/IntelRealSense/realsense-ros/issues/1515#issuecomment-744223832

As advised by Yusheng in the image quote that you kindly provided, a source code build of librealsense will not be based on RSUSB backend by default.

DeboBurro, 2022-10-17:
hi @MartyG-RealSense, thank you so much. We were also suggested to use the branch `development` to build the kernel modules based on the discussion in the [PR](https://github.com/IntelRealSense/librealsense/pull/10625).
So I ran through the following commands and it builds the patch kernel modules successfully.
1. `git clone -b development https://github.com/IntelRealSense/librealsense.git`
2. `cd librealsense`
3. `./scripts/patch-realsense-ubuntu-lts-hwe.sh`

The result shown below says successfully built and installed but I am a little bit concern about some `*.ko` files related to `v4l2` were skipped.
```
      .
      .
Skipping BTF generation for /tmp/librealsense/ubuntu-focal/drivers/media/v4l2-core/videobuf-core.ko 
due to unavailability of vmlinux
  BTF [M] /tmp/librealsense/ubuntu-focal/drivers/media/v4l2-core/v4l2-dv-timings.ko
Skipping BTF generation for /tmp/librealsense/ubuntu-focal/drivers/media/v4l2-core/v4l2-dv-timings.k
o due to unavailability of vmlinux   
  LD [M]  /tmp/librealsense/ubuntu-focal/drivers/media/v4l2-core/videodev.ko
  LD [M]  /tmp/librealsense/ubuntu-focal/drivers/media/v4l2-core/videobuf-vmalloc.ko
  BTF [M] /tmp/librealsense/ubuntu-focal/drivers/media/v4l2-core/videobuf-vmalloc.ko
Skipping BTF generation for /tmp/librealsense/ubuntu-focal/drivers/media/v4l2-core/videobuf-vmalloc.
ko due to unavailability of vmlinux  
  BTF [M] /tmp/librealsense/ubuntu-focal/drivers/media/v4l2-core/videodev.ko
Skipping BTF generation for /tmp/librealsense/ubuntu-focal/drivers/media/v4l2-core/videodev.ko due t
o unavailability of vmlinux
make: Leaving directory '/tmp/librealsense/ubuntu-focal'
Patched kernels modules were created successfully

[sudo] password for burro:
Replacing videodev  -
        Applying the patched module ...  succeeded
Replacing uvcvideo  -
        Applying the patched module ...  succeeded
Replacing hid_sensor_accel_3d  -
Module is resident, unloading ...  succeeded.
        Applying the patched module ...  succeeded
Replacing hid_sensor_gyro_3d  -
Module is resident, unloading ...  succeeded.
        Applying the patched module ...  succeeded

Script has completed. Please consult the installation guide for further instruction.
```

MartyG-RealSense, 2022-10-24:
Hi @DeboBurro and @fiorano10  Do either of you require further assistance with this case, please?  Thanks!

closed 2022-10-24 by DeboBurro