realsenseai/librealsense #10989: How can I install for Nvidia AGX Xavier with Jetpack 5?

DaveBGld, 2022-10-13:
I need to migrate my ROS realsense application to an AGX Xavier with Jetpack 5 installed

I tried installing using vcpkg following the instructions [here](https://github.com/IntelRealSense/librealsense)
```

git clone https://github.com/Microsoft/vcpkg.git
cd vcpkg
./bootstrap-vcpkg.sh
./vcpkg integrate install
./vcpkg install realsense2
```

**After completing those commands without errors, I could not find any of the command line tools such as `rs-enumerate-devices` and `realsense-viewer`.**

This is the log for `./vcpkg install realsense2` : [vcpkg build of realsense2 log.txt](https://github.com/IntelRealSense/librealsense/files/9772473/vcpkg.build.of.realsense2.log.txt)

Then I tried downloading the code and building it as described [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)

However, when I run the command

`./scripts/patch-realsense-ubuntu-lts.sh`

I get an error 

**> Unsupported kernel version 5.10.104-tegra . The Focal patches are maintained for Ubuntu LTS with kernel 5.4, 5.8, 5.11 only**

If Jetpack 5 is not supported, the the relevant question is **What is the latest version of Jetpack supported?**

**And, What is the proper install of the realsense ROS package?**

Many thanks!

Dave

MartyG-RealSense, 2022-10-13:
Hi @DaveBGld  The instructions at the **distribution_linux.md** page are intended for computers with x86 / x64 CPU architecture such as desktop and laptop PCs.  Nvidia Jetson boards have their own dedicated installation instructions and patch script at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

The most recent support, which was added in SDK 2.50.0, is for L4T 32.6.1 / JetPack 4.6.

There have been RealSense users who have tried JetPack 5 and whilst it partially worked for them, there could be problems such as the IMU on IMU-equipped RealSense camera models being inaccessible.

In regard to installing the ROS wrapper on Jetson, I recommend first installing librealsense by whichever method works best for you, and then building the wrapper from source code afterwards.

DaveBGld, 2022-10-13:
After researching and trying, it turns out that in order to install Jetpack 4.6  I need to reformat my workstation and install Ubuntu 18.04 down from my current 20.04!!! That will shot me down for a couple of days!!! Maybe 3 to install all the stuff I use...  Plus another to install the jetpack and all the toolchain needed including realsense2. And I already have a couple of days in trying to figure out the problems installing realsense2... 

Looking for a chill pill....

MartyG-RealSense, 2022-10-14:
In https://github.com/IntelRealSense/librealsense/issues/10416 two RealSense users with Xavier AGX and NX were able to use 20.04 and JetPack 5 with librealsense successfully if they built librealsense from source code.  The NX user successfully tested ROS1 Noetic with it.

MartyG-RealSense, 2022-10-20:
Hi @DaveBGld  Was the comment above helpful to you, please?  Thanks!

DaveBGld, 2022-10-20:
Hi @MartyG-RealSense . I have mixed feelings. Jetpack 5 was out in April, and yet Intel has not really made the effort to make the realsense supporting software available, 7 months later? I am developing a professional application and don't have room fore **hacked solutions.**  

I know Intel and Nvidia are competitors and as such behave as enemies, but us the users of your technology suffer in the end with this behavior...  I have wasted about a week of my time on this issue which delayed my whole team. 

So, @MartyG-RealSense , while you were helpful as a supporting member of the Intel staff, the fact that I was sent on this bad trip, is on Intel policies... And THAT was not helpful.

Thanks!!

closed 2022-10-20 by DaveBGld

MartyG-RealSense, 2022-10-20:
Thanks very much for the update.  I'm sincerely sorry that you experienced difficulties with JetPack compatibility.