realsenseai/librealsense #10991: Question about sync when sensor API is used

Marenix, 2022-10-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.13.0.50 | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |                                   | 
| Platform                                 | PC  |
| SDK Version                            |  2.50.10  | 
| Language                            |                   | 
| Segment			|                        | 

Hi, I have a question about sync.cpp. I've been looking into it a bit, and I've noticed something strange when using sensor API instead of pipeline API. So, when using sensor API, when the first frame arrives, the following is done:

[image]

In line 197, we use "dev->create_matcher(frame)" to create all matchers we need (which in the end goes to the composite_matcher constructor. What I'm interested in is the callback that's set to the matcher. From what I can gather, this callback is called when the matchers previously created with "create_matcher()" manage to sync frames. So, when frames are synced, the callback set here should be called, and it's currently set so that it, again, calls sync with the fameset that was already synced. 

What I believe the callback should be is following:

[image]

This has been taken from "syncer-processing-block.cpp" and this is the callback that's set when pipeline API is used. It basically moves the composite frame into queue which is then used for whatever is next (in the end being outputed when "wait_for_frames()" is called. 

The only reason this is currently working is because when sync is used the frame is outputed if there are no missing streams which will happen sooner or later. Can I please get some information if this is deliberate, and if so, what the reasoning behind it was. 

Thank you

MartyG-RealSense, 2022-10-13:
Hi @Marenix  I will discuss your question with my Intel RealSense colleagues.  Thanks very much for your patience!

MartyG-RealSense, 2022-10-20:
Hi @Marenix  My colleagues researched your question and determined that the behavior is by design, as described by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/5675#issuecomment-576018859

The syncer will only output a frameset only when the matched frames are found.  The pipeline will output a frameset when one of the streams is not available.

Marenix, 2022-10-25:
Hi @MartyG-RealSense, thank you for the answer. In the next few days I'll do a few tests based on your answer and then get back to you to close the issue if everything works as expected.

MartyG-RealSense, 2022-10-25:
You are very welcome.  I look forward to your next update.  Good luck with your tests!

MartyG-RealSense, 2022-11-02:
Hi @Marenix  Do you have an update about this case that you can provide, please?  Thanks!

Marenix, 2022-11-04:
Hi @MartyG-RealSense I am currently working on some other things, so I still haven't had a chance to do any tests on this. Please give me until the end of next week, and I'll try to get to it as soon as I possibly can.

MartyG-RealSense, 2022-11-04:
That's no problem at all.  Good luck with your work and thanks again for the update!

Marenix, 2022-11-23:
Hi @MartyG-RealSense, sorry it took me so long to get back to this. I've ran some tests with both the original callback and with the changed one (both which I mentioned in the first post). I haven't noticed any change when I changed the callback. 

Just for clarification, while using syncer, I should never receive 2 framesets which have one of the frames identical in both framesets, right? Well, I encountered that case (where I have 2 framesets in which I have the same depth or color frame while the other is different) while using both the original callback and the one I suggested. If this shouldn't happen, is there any simple example with which I could check if there might be some problem in my application?

Thank you

MartyG-RealSense, 2022-11-24:
A RealSense team member advises at https://github.com/IntelRealSense/librealsense/issues/5107#issuecomment-546870736 that use of **poll_for_frames()** avoids duplicate frames.

Marenix, 2022-11-30:
Hi @MartyG-RealSense, after some additional tests, I found an issue in my example. I have since fixed the issue and now I do not see duplicated frames with wait_for_frames() (haven't tried poll_for_frames() after fixing the problem).

Going back to my original post, I now tried the example with both version of the callback I mentioned there, and I do not see any difference in my example. Neither version gives a duplicate of the previous frame if the new one is missing. So, looking back to your original answer, I cannot see the syncer giving unmatched frames with either callback. Could there be some other reason for choosing one callback over the other?

MartyG-RealSense, 2022-12-01:
My honest answer to your question of whether there could be some other reason for choosing one callback over the other is that I do not know.  I do apologize.

MartyG-RealSense, 2022-12-10:
Hi @Marenix  Bearing in mind the comment above, do you require further assistance with this case please?  Thanks!

Marenix, 2022-12-12:
Hi @MartyG-RealSense Thanks for the clarifications you have given me. Since there is nothing further that can be said on the topic, I think it can be closed

MartyG-RealSense, 2022-12-12:
You are very welcome.  As you suggest, I will close the issue.  Thanks again!

closed 2022-12-12 by MartyG-RealSense