realsenseai/librealsense #10993: Coordinate System of d435

Akumar201, 2022-10-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D450 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.14.0-1052-oem | 
| Platform                                 | laptop.  |
                     | 

### Issue Description
Hello, I would like to know what is the Coordinate System frame that is being used in Realsense d435 camera. I found something relative but would like get valid information. Thank you 

[image]

MartyG-RealSense, 2022-10-13:
Hi @Akumar201  The RealSense SDK's **Projection** documentation at the link below is an introductory reference to RealSense coordinate systems.

https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20

I recommend Intel's **Projection, Texture Mapping and Occlusion** white-paper guide for further reading on the subject.

https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras

Akumar201, 2022-10-14:
I resolved the issue. Please refer to the below image for 
[image]

Tracking System Coordinate System for T265
[image]

closed 2022-10-14 by Akumar201

MartyG-RealSense, 2022-10-14:
That's excellent news that you found the information that you needed.  Thanks very much for the update!