realsenseai/librealsense #10994: Camera Calibration using easy handeye for d435

Akumar201, 2022-10-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D450 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.14.0-1052-oem | 
| Platform                                 | laptop.  |
                     | 

I am doing camera calibration using easy handeye(https://github.com/IFL-CAMP/easy_handeye) package. After doing camera on base I found out the parameters to be 

> `parameters:
> eye_on_hand: false
>   freehand_robot_movement: false
>   move_group: xarm6
>   move_group_namespace: /
>   namespace: /xarm_realsense_handeyecalibration_eye_on_base/
>   robot_base_frame: link_base
>   robot_effector_frame: link_eef
>   tracking_base_frame: camera_color_optical_frame
>   tracking_marker_frame: camera_marker
> transformation:
>   qw: 0.06571339891764572
>   qx: -0.06259327310419002
>   qy: -0.7003283367747712
>   qz: 0.7080282847987625
>   x: 0.2979904862894281
>   y: 1.673693666442873
>   z: 0.971493072034135

 `

After manually checking and following the convention for translation in x,y,z were almost same. 
However, the orientation is getting different as in y seems to be correct but x and z are interchanged(when I am assuming taking the values of x and z then it make sense).

I would like to share the resources that I use for coordinate system 

[image]

[image]

Akumar201, 2022-10-14:
I resolved this issue, 
So, treat the robot base as (0,0,0) , remember front is roll, side is pitch and up is yaw. Front can be defined as the side which is facing the camera. Also find out the right coordinate system for your camera for my case it was d435

[image]
where green is y, red is x and z front axis in blue. Then try to rotate the base frame by any of roll, pitch or yaw to get the same orientation as the camera. Then note down the angles for roll, pitch and yaw that can be compared to the result in easy hand-eye calibration. The quaternions in the yaml are in radians, you can use the following code to convert them into euler form in degree

```
import math
 
def euler_from_quaternion(x, y, z, w):
        """
        Convert a quaternion into euler angles (roll, pitch, yaw)
        roll is rotation around x in radians (counterclockwise)
        pitch is rotation around y in radians (counterclockwise)
        yaw is rotation around z in radians (counterclockwise)
        """
        t0 = +2.0 * (w * x + y * z)
        t1 = +1.0 - 2.0 * (x * x + y * y)
        roll_x = math.atan2(t0, t1)
     
        t2 = +2.0 * (w * y - z * x)
        t2 = +1.0 if t2 > +1.0 else t2
        t2 = -1.0 if t2 < -1.0 else t2
        pitch_y = math.asin(t2)
     
        t3 = +2.0 * (w * z + x * y)
        t4 = +1.0 - 2.0 * (y * y + z * z)
        yaw_z = math.atan2(t3, t4)
     
        return roll_x, pitch_y, yaw_z # in radians # roll and pitch can be diffrent in your case so choose respectively

val = euler_from_quaternion(-0.06259327310419002,-0.7003283367747712,0.7080282847987625,0.06571339891764572)
degrees = []
for i in range(len(val)):
    degrees.append(math.degrees(val[i]))
print(val)
print(degrees)
```

closed 2022-10-14 by Akumar201

MartyG-RealSense, 2022-10-14:
Thanks very much @Akumar201 for the update and for sharing your solution with the RealSense community!