realsenseai/librealsense #10996: Recording into sequential .bag files with recorder

nixone, 2022-10-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Operating System & Version |   Win 10  | 

| Platform                                 | PC |
| SDK Version                            |  2.51.1                       | 
| Language                            |  python                   | 

### Issue Description

Hi, is it possible to record (and in the same time use) the stream from a camera and save the recording into subsequent files? 

I tried two different approaches:
- Every time, when I want to split the .bag file (let's say every few minutes), I stop the pipeline, reconfigure with `cfg.enable_record_to_file`. This works well, but unfortunately stopping the pipeline and restarting it is fairly slow, the camera doesn't return any meaningful frames for several seconds.
- Creating new `rs2::recorder` every time when I want to split the .bag file while keeping the pipeline running. Unfortunately, this approach works only for the first .bag file. When I re-create a new `rs2::recorder` object with the same reference to a device but a new file name, the new `.bag` file is no longer recorded (it is created, it only has cca 72 KB of data). I assume the problem is that the original `rs2::recorder` object is still somehow tied with the `rs2::device` object, yet there is no possibility for me to close or release the original `rs2::recorder`.

**Does anyone have an idea how I could solve this?**

Essentially, my problem is as this: I want to use the data for some product evaluation, but while that evaluation is running, I want to record the same data in the background. I want to make sure we have enough storage on the device, so we need to split the data into multiple files per several minutes. We are deleting the oldest files, so we can always keep some history.

Thanks anyone for any tips!

MartyG-RealSense, 2022-10-14:
Hi @nixone  The librealsense SDK does not have a feature for splitting bag recording into new files automatically based on recording duration or file size.  This feature is available in the RealSense ROS wrapper via ROS's **rosbag record** feature and its **--split** function - as described at https://github.com/IntelRealSense/librealsense/issues/2424#issuecomment-423284173 - though this may not be suitable for your particular project if it has to be a Python application.

An instruction that librealsense does have that is supported in Python is **save_single_frameset()**, which saves every frame as an individual bag file instead of creating a single long recording.  

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.save_single_frameset.html

A RealSense user created a complete Python script specifically for this instruction at the link below.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/single_frameset.py

https://github.com/IntelRealSense/librealsense/issues/10941 is another Python example of its use.

Alternatively, a Python approach suggested at https://github.com/IntelRealSense/librealsense/issues/4801 for a RealSense user with a similar situation to yours of not wanting to reset the pipeline was to pause and unpause recording instead.

MartyG-RealSense, 2022-10-20:
Hi @nixone  Do you require further assistance with this case, please?  Thanks!

nixone, 2022-10-20:
I think we can close the issue for now, you provided very good tips, thanks for that! :)

closed 2022-10-20 by nixone